justcallmekoko/ESP32Marauder #153: [WIKI CONTRIBUTION] - Update Page -> [Arduino IDE Setup]

641i130, 2022-09-14:
5. With any text editor, open: 
`C:\Users\<USERNAME>\AppDate\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1\platform.txt`
or on linux: 
`/home/<USERNAME>/.arduino15/packages/esp32/hardware/esp32/1.0.6/platform.txt`
6. Add `-w` to the end of line `33` so it appears like so
        `build.extra_flags.esp32=-DARDUINO_SERIAL_PORT=0 -w`
or on Linux:
        `build.extra_flags=-DESP32 -DCORE_DEBUG_LEVEL={build.code_debug} {build.defines} -w`
7. Add `-zmuldefs` to the end of line 27 (`compiler.c.elf.libs.esp32=` or `compiler.c.elf.libs`)

deadloko, 2022-10-13:
Also, it is worth noting, that for esp32s2 based device, user should modify esp32s2 section in platform.txt. And, if arduino ide is open, settings will be applied only after restart (atleast on Windows).

HoustonAsh, 2023-04-25:
For me on Manjaro linux (kernel 6.1.25-1):
`~/.arduino15/packages/esp32/hardware/esp32/2.0.8/platform.txt`
`151.  build.extra_flags=-DESP32 -DCORE_DEBUG_LEVEL={build.code_debug} {build.loop_core} {build.event_core} {build.defines} {build.extra_flags.{build.mcu}} -w`

And

`109. compiler.c.elf.libs={compiler.c.elf.libs.{build.mcu}} -zmuldefs`