justcallmekoko/ESP32Marauder #154: Allow for deauth only a specific mac address

Frisch12, 2022-09-18:
As some devices (e.g. every Apple device) ignores broadcast deauth packages it would be nice if there would be a function to target a specific or all clients connected to a wifi network.

6q6, 2022-09-21:
Devices ignore deauths when management frames are enabled (mandatory/optional), this is configured from the AP itself.
The esp32 Marauder deauth attack targets the AP, resulting in deauthenticating all connected clients w/o management frames. 

Specific client targeting when they're already connected to an AP (assuming you have no info) is something that's impossible due to the absence of probe requests, which are only sent when clients are disconnected.

justcallmekoko, 2022-11-03:
Implemented in v0.9.17