justcallmekoko/ESP32Marauder #156: Board will not run Marauder unless I flash it via OTA, upon reboot it will not run it anymore

noodleboy91, 2022-09-24:
I flashed my ESP32 with the "old_hardware" firmware using OTA update. It works fine after flashing until I reboot the board, then it will not scan ap(s) or perform any other Marauder function. It appears to be going straight into OTA update mode, as I can see the Marauder OTA ap in my ap list  on computer (not sure if this is typical when it is running normally), any ideas of what might be going on?

For reference, this is the board I am using: https://www.amazon.com/gp/product/B07V42CJRF

Thank you

SharpEdgeMarshall, 2022-10-07:
Same issue here, the automatic reboot is good and then reverts to Marauder OTA
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4

Marauder OTA Update

Configuring update server...
Initializing WiFi...
Starting softAP...

Displaying settings to TFT...
IP address: 192.168.4.1
Setting server behavior...
Finished setting server behavior
Starting server...
Completed update server setup

---------------------------------

SSID: MarauderOTA
Network Password: justcallmekoko
Username: admin
Password: admin
Update server: Client connected -> 1
onJavaScript(void)
onJavaScript(void)
onJavaScript(void)
onJavaScript(void)
Update: esp32_marauder_v0_9_15_20220907_old_hardware.bin
Update Success: 1553680
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
ESP-IDF version is: v4.4-dev-2313-gc69f0ec32
Failed to mount SD Card
SD Card NOT Supported
CLI Ready

              @@@@@@                        
              @@@@@@@@                      
              @@@@@@@@@@@                   
             @@@@@@  @@@@@@                 
          @@@@@@@      @@@@@@@              
        @@@@@@            @@@@@@            
     @@@@@@@                @@@@@@@         
   @@@@@@                      @@@@@@       
@@@@@@@              @@@@@@@@@@@@@@@@       
@@@@@                 @@@@@@@@@@@@@@@       
@@@@@                   @@@@@@@             
@@@@@                      @@@@@@           
@@@@@@                       @@@@@@@        
  @@@@@@                        @@@@@@@@@@@@
    @@@@@@@                          @@@@@@ 
       @@@@@@                     @@@@@@    
         @@@@@@@                @@@@@@      
            @@@@@@           @@@@@@         
              @@@@@@@      @@@@@@           
                 @@@@@@ @@@@@@              
                   @@@@@@@@@                
                      @@@@@@                
                        @@@@                

--------------------------------

         ESP32 Marauder      

            v0.9.15

       By: justcallmekoko

--------------------------------

> ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4

Marauder OTA Update

Configuring update server...
Initializing WiFi...
Starting softAP...

Displaying settings to TFT...
IP address: 192.168.4.1
Setting server behavior...
Finished setting server behavior
Starting server...
Completed update server setup

---------------------------------

SSID: MarauderOTA
Network Password: justcallmekoko
Username: admin
Password: admin

```

frenez, 2022-10-08:
Hi, i sort of solved the problem with erasing the marauder firmware on this devboard, I did everything as in the OTA installation instructions up to step 9, and instead of installing the bin file via wifi, I used the method via the esp flash downloader tool https://github.com/SequoiaSan/Guide-How-To-Upload-bin-to-ESP8266-ESP32 , now the firmware does not crash when the power is turned off or rebooted , however, since our devboard version is esp_wroom_32, some functions will not work, an esp32s2 devboard is needed.

SharpEdgeMarshall, 2022-10-10:
For me the fix was to use exactly arduino ESP `2.0.0-rc1` instead of latest version

noodleboy91, 2022-10-12:
Thank you both for the suggested fixes, I am going to give it a try :)

joeldg, 2022-12-02:
Same issue here, I tried all kinds of things and just ended up using the EasyInstall.py script... which works, but.. it's not OTA

djecom1, 2022-12-13:
> 

Thank you ! It works for me and my ESP32 WROOM.

djecom1, 2022-12-13:
> Hi, i sort of solved the problem with erasing the marauder firmware on this devboard, I did everything as in the OTA installation instructions up to step 9, and instead of installing the bin file [...]

Thank you, it works for my ESP32 WROOM.

Chaingum, 2023-02-07:
I had the same problem and found some kind of workaround by chance, my setup is a ESP-WROOM-32 in a LOLIN32 V1 board.
In the Flipper app I started the ota update (see screen) and updated again with the browser and it worked.
I can now restart the esp and it boot on the Marauder firmware
[image]

csillag13, 2023-02-16:
> I had the same problem and found some kind of workaround by chance, my setup is a ESP-WROOM-32 in a LOLIN32 V1 board. After a successful OTA flash, without resteting the board, in the Flipper app I started the ota update (see screenshot below) and flashed again with the browser and it worked. I can now restart the esp and it boot on the Marauder firmware
> 
> [image]

M8 you saved my day, gj, worked for me too!

Poughk33spie, 2023-05-21:
> M8 you saved my day, gj, worked for me too!

how do you get the ota update thing