justcallmekoko/ESP32Marauder #166: Flipper Zero Marauder (dev board) Unable to Deauth and then send beacon frames

Archimedes127, 2022-11-07:
Hardware and firmware
Flipper Zero - Unleashed 
FZ Dev Board - Marauder v0.9.17

The dev board running marauder requires resetting after sending deauth frames and switching to beacon frames.  If the board is not reset, the flipper dev board does show that it sending beacon frames (led indicator turns red) however utilising a mobile device to scan for the beacons brings back no results. If the board is reset and beacon frames are re-sent the beacons are visible after scanning.  

Steps to reproduce the behavior:
1. Go to "SSID" <add name> create the name of required ssid
2. Go to scanap
3. Start scanap, ap's are visible and stored in list 
4. Open AP list. Confirm AP 
5. Select AP via "select" 
6. Carry out "attack" (deauth) on selected ap
7. Stop attack
8. Go to "Beacon Spam" <ssid list> click to start. 
9. No beacons are displayed on WiFi monitoring device (in this case mobile phone) 

**Expected behavior**
A deauth attack followed by a beacon spam afterwards without having to reset the devboard between changing.  

**Screenshots**

[image]

[image]

[image]

[image]

No ssid shown when looking for ap's via phone (reset of board required, and it is possible) 

[image]

After emulating the issue, it was noticed that no further actions can be taken without resetting the devboard. The initial deauth attack appears to stop further pen testing.

Archimedes127, 2022-11-07:
Double posted deauth pic. Should've been beacon spam (cut me some slack, jetlag is real)

[image]

Archimedes127, 2022-11-07:
After a deauth is carried out, if you try to attack via Rickroll, no beacons appear. Reset the board and carry out the same attack, they show as intended.

justcallmekoko, 2022-11-07:
I am unable to reproduce the bug myself. I followed this order of operations to reproduce the issue on my end. Please let me know if there was something I misunderstood.

1. `scanap`
2. `stopscan`
3. `ssid -a -g 3`
4. `select -a <index of target ap>`
5. `attack -t beacon -l` (first beacon spam)
6. `stopscan`
7. `attack -t deauth`
8. `stopscan`
9. `attack -t beacon -l`
10. `stopscan`
11. `attack -t rickroll`
12. `stopscan`

All beacon spam attacks following the deauthentication attack executed as expected. This was tested on a Marauder Mini and Flipper Zero Dev Board as the transmitter. I monitored the attacks with a separate Marauder. All of the attacks were detected by the monitoring Marauder.

Archimedes127, 2022-11-07:
The only hardware variation is that I'm using a flipper zero/ Flipper dev board and not a marauder mini/flipper dev board as you have done above.

There must be an issue with the interface between the flipper zero and the devboard possibly. The WiFi dev board shows activity when initialised via the flipper zero (scan and attack) with green and red LEDs respectively.

justcallmekoko, 2022-11-07:
Oh no when I say I tested this with a mini and a flipper dev board, I meant they were operating independent of each other. The flipper dev board was plugged into the flipper zero and the Marauder Mini was operating as a standalone device. Both devices performed the list of commands above and the output from both was as expected. I will need to see if there is something else I need to do to get the same thing to happen

Archimedes127, 2022-11-07:
Can you confirm the fw versions on the FZ and Dev board? Cheers

justcallmekoko, 2022-11-07:
I am currently running v0.9.17 on both the Marauder Mini and the Dev Board. On the FZ, I am running unleashed version unlshd-012

Archimedes127, 2022-11-08:
I'm running unlshd-014.

I'll roll back to 012 and see if I can emulate the issue.

justcallmekoko, 2022-11-27:
Any updates on this issue?

Archimedes127, 2022-11-27:
None so far. Some other users have reported the same thing in some groups I'm in.  I'm trying to figure out if there's any settings/hardware variations (sd card, general flipper settings) but in all honesty I've been busy with work, so have slacked a bit regarding it.

Wh331z, 2022-12-02:
I have the exact same issue as above, I am using the 0.9.1.7 Marauder OTA flashed to the flipper 0 wifi dev board, and I'm running unleashed-0.1.4. So I'm not sure if I'm doing something wrong or there's a bug in the firmware. 

Cheers

Archimedes127, 2022-12-02:
I've ordered another dev board (dedication to the cause) to rule out it possibly being the board. I'm also going to get another FZ to rule out possible hardware issues. I expect delivery in the next week so will keep you posted.
I'm leaning towards the unleashed firmware, but as koko can't emulate the issue it almost seems like it's an intermittent thing with maybe different hardware batches? Guess we'll find out...

justcallmekoko, 2022-12-23:
@Archimedes127 Please update to v0.10.0 when you get the chance and check if the issue is still present

Archimedes127, 2022-12-23:
Will do mate.

Archimedes127, 2022-12-27:
Updated and issue is fixed! Good job.

Archimedes127, 2022-12-27:
So it was an issue with older versions. Good job fixing it justcallmekoko, cheers!

closed 2022-12-27 by Archimedes127