justcallmekoko/ESP32Marauder #173: Marauder Mini crash/reboot on 'Select APs'

D4DB0DY, 2022-11-09:
**Describe the bug**
Marauder mini crashes and reboots when opening the "select AP's" menu. This only happens if the device has found many AP's during the scan. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Wifi', 'sniffers', 
2. Click on 'scan APs' (let it scan all channels)
3. Go to 'Wifi', 'general'
4. Click on 'Select APs'
5. Menu will open for about 1 second and start showing scanned AP's then will crash and reboot the device. 

**Expected behavior**
I would expect to be able to scroll down and select APs to target with an attack.

**Screenshots**
Video attached.

**Marauder (please complete the following information if applicable):**
 - Firmware version: v0.9.17
 - Hardware version: Marauder Mini

**Desktop (please complete the following information if applicable):**
 - OS: N/A
 - Browser N/A
 - Version N/A

**Smartphone (please complete the following information if applicable):**
 - Device: N/A
 - OS: N/A
 - Browser N/A
 - Version N/A

**Additional context**
Note: this doesn't happen if I only let it scan 4 or so AP's.

https://user-images.githubusercontent.com/106566379/200716885-6182fd64-7c94-489e-9d05-d403e9787999.mp4

justcallmekoko, 2022-11-17:
This will be resolved in v0.9.18

justcallmekoko, 2022-11-22:
https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.9.18

closed 2022-11-22 by justcallmekoko