justcallmekoko/ESP32Marauder #178: Diy marauder mini

skyrim55, 2022-11-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

skyrim55, 2022-11-19:
Sir, with due respect i really love this project and respect ur work i just wanna recreat this project if is it possible to help me provide some information like what display used and some wirring diagram or schematic of "marauder mini" that caught my interest

trisp3ar, 2022-11-19:
For what you are looking for, please see the [Marauder Mini Repo](https://github.com/justcallmekoko/MarauderMini).
It contains most of the information you are looking for. If you require schematics, please let us know. @justcallmekoko hasn't added them yet and it's prob something that has to be done in the future.
Please also use the issue form correctly.

skyrim55, 2022-11-21:
Thanks a lot

closed 2022-11-21 by skyrim55