justcallmekoko/ESP32Marauder #198: is it possible to implement adafruit esp32-s2 WiFi QT Py?

pragamichele, 2022-12-22:
this board is very small with built in wifi, but it doesn't seem compatible with flipper zero

justcallmekoko, 2022-12-23:
It is possible. You would need to build the firmware from source and configure specifically for whatever connections the dev board makes

arag0re, 2023-06-17:
@justcallmekoko i got the fimware running on the qt py esp32s2, where do i find the pinout part for rx/tx pins ?

arag0re, 2023-06-29:
> @arag0re Did You flashed with FZEasy or Spacehuhn Web? Did You build it form sources or You used already compiled version ? Mine gets flashed but I cannot get Marauder CLI via USB Serial connection . Any hints highly appreciated since this board would be very cool together with F0 + Antenna ;)  

I compiled it from source and flashed it through arduino-ide 1.8. 
Im successfully gettung usb-serial output ans can run "scanap" via cli. And it actually finds wifis. So that part is working for me. I am unable to get it to work with my flipper tho. I changed pinout in sourcecode but not getting any output on those rx:tx pins

arag0re, 2023-06-29:
Also setting up the Arduino-IDE beforehand is very important, like described [here](https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup).
Also u need to upload the Minimal SPIFFS stuff first and then use the regular "upload" second.

[image]
[image]

Just read through the guide. Its really detailed and worked for me the first time.

xjazzfreakx, 2023-06-29:
@arag0re Thanks for the tips ! Now I have to take some time and do the compile -path ;) Then check if the pins are delivering. Did You tried it with UART console app ?

arag0re, 2023-07-06:
I got a working version, not completly perfect but good enough

arag0re, 2023-07-09:
https://github.com/arag0re/ESP32Marauder

arag0re, 2023-07-09:
@xjazzfreakx @pragamichele