justcallmekoko/ESP32Marauder #202: LED of FlipperZero WiFi DevBoard not working with v2.0

bkimminich, 2022-12-31:
**Describe the bug**

After flashing with [FZ_Marauder_v2.0.zip](https://github.com/UberGuidoZ/Flipper/blob/main/Wifi_DevBoard/FZ_Marauder_Flasher/FZ_Marauder_v2.0.zip) the LED of the WiFi DevBoard would no longer flash except for a short blink when plugging the board into the FZ.

Running AP scans or DeAuth attacks would not cause the LED to shine in the intended color. 

When I downgraded the Marauder firmware to v1.7 the LED worked normally again in any mode and with the expected color.

marcoboehrs, 2023-01-17:
Led is working again on my side. What I did in the marauder app: settings: restore -> settings: EnableLed. I think the config file was missing, because after selecting settings: restore a new one was created

bkimminich, 2023-05-18:
This works! ❤️

closed 2023-05-18 by bkimminich