justcallmekoko/ESP32Marauder #204: esp32marauder 4.5 wont start

johnjohnsp1, 2023-01-07:
Hi, just received my new esp32marauder v4.5, bought a battery 1500 mAh +3.7V but once i plugged in nothin happen, no video but also no boot or any led turn on , for example the indicator of the battery. nothin happen also if i plug in the usb-c cable. any idea ? is new device damaged physcally and wont start ?

thanks
[image]

justcallmekoko, 2023-01-17:
Post a picture of the back of the unit with the battery connected.

johnjohnsp1, 2023-01-17:
hi,

￼￼

need anything else ?
let me know

thanks

justcallmekoko, 2023-01-24:
I'm not sure what the small graphic is in your most recent message. I just see "￼￼"

I just need to see the battery connector on the back of the PCB and how the battery is connected to it

grendel6point5, 2023-04-30:
i bought a marauder and it turned on once with the usb cord powering it but now nothing happens, even the leds arent turning on. i tested the usb power and its good . the marauder gets hot so its obviously got power going to it. is there anything i can try to get it to work again?