justcallmekoko/ESP32Marauder #219: Evil Twin Attack

sealldeveloper, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
No it is not related to a problem, rather a request for an option to use an Evil Twin attack.

**Describe the solution you'd like**
An option to use an Evil Twin attack, if possible

**Describe alternatives you've considered**
None

**Additional context**
I'm quite new to the ESP32 and its capabilities so I am unaware if its possible, but it would be a nice feature.

SantiEich, 2023-04-03:
I'm trying to do that, did you continue to investigate if it's possible?

sealldeveloper, 2023-04-04:
No, I'm not particularly proficient in this aspect of development but it is a feature I would like.

Moicky, 2023-04-22:
Any progress? Would be interesting if there's already an existing solution for the Flipper Zero. The functionality should be possible using a custom firmware ([this one for example](https://github.com/DarkFlippers/unleashed-firmware)) with the preinstalled ESP32 Marauder?

SantiEich, 2023-04-24:
Nothing new for now

nikoli0218, 2023-07-02:
Mauraudercentauri has the capability not sure if that ports to the flipper or not

SantiEich, 2023-07-25:
Something good here https://github.com/bigbrodude6119/flipper-zero-evil-portal/tree/6b87f667ba9669c386fe00454bbef0ded7e34af1 
@bigbrodude6119 👏