justcallmekoko/ESP32Marauder #224: ESP32 will not boot up after failing to reboot post firmware update

eltee3, 2023-01-31:
**Describe the bug**
ESP32 will not boot up after failing to reboot post firmware update

**To Reproduce**
Steps to reproduce the behavior:
1. Web browser firmware update completed
2. says: rebooting -- did not reboot automatically
3. Manually powered off
4. Does not reboot

**Expected behavior**
The device to boot up after firmware update

**Marauder (please complete the following information if applicable):**
 - Firmware version:  v1.10.23

Brutal-Pidgeon, 2023-04-04:
I got the same issue and I thought I messed up everything by myself, but now I see I am not the only one who got this problem.

eltee3, 2023-05-17:
No resolution as of yet.

CyberTechNex, 2024-05-09:
Hi if there is still no resolution try using the bin file named old hardware. That fixed the issue and it rebooted perfectly fine.

Brutal-Pidgeon, 2024-05-10:
> Hi if there is still no resolution try using the bin file named old hardware. That fixed the issue and it rebooted perfectly fine.

solved this issue months ago. I can confirm your solution will work