justcallmekoko/ESP32Marauder #226: MarauderOTA Flipper Wifi Dev Board Compile Error

ryangott, 2023-02-03:
**Describe the bug**
I get to step 7 of [Installing Firmware via OTA](https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-via-ota) and get a compiling error for the wifi dev board.

**To Reproduce**
Steps to reproduce the behavior:
Follow the instructions under [Arduino IDE Setup](https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup)
Open the Arduino IDE
Plug your ESP32 into a USB port and select the COM port under Tools>Port
You may need to hold the "boot" button while inserting the USB
Select the appropriate ESP32 module under Tools>Boards
Select LOLIN D32 for the following:
Marauder v4 (OG)
Marauder v6
Marauder Mini
Select Adafruit ESP32 Feather for the following:
Marauder Kit
Select ESP32-S2 dev board for the following:
Marauder Flipper (Flipper Zero WiFi Dev Board)
If you do not see this board option, ensure your Arduino IDE is on the latest version
For partition scheme, select Minimal SPIFFS (Large APPS with OTA)
For Marauder Flipper, select 16MB for Flash Size
Copy the ESP32 Marauder OTA Upload firmware source from [here](https://raw.githubusercontent.com/justcallmekoko/ESP32Marauder/master/MarauderOTA/MarauderOTA.ino) into your Arduino IDE window

**Expected behavior**
MarauderOTA sketch sent to flipper wifi dev board and compiled successfully. 
**Screenshots**
[image]

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.9.17]
 - Hardware version: [e.g. OG] 

**Desktop (please complete the following information if applicable):**
 - OS: MacOS Monterey 12.5
 - Browser: Chrome
 - Version 109.0.5414.87

**Smartphone (please complete the following information if applicable):**
 - Device: n/a
 - OS: n/a
 - Browser n/a
 - Version n/a

**Additional context**
Error Message: 
Arduino: 1.8.16 (Mac OS X), Board: "ESP32S2 Dev Module, UART0, UART0, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 16MB (128Mb), 921600, None"

Multiple libraries were found for "WiFi.h"
 Used: /Users/ryangott/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/libraries/WiFi
 Not used: /private/var/folders/sd/ygr2hj214690s33167syz0v80000gn/T/AppTranslocation/ED7B9D9C-82D2-43A3-9009-FD079566B128/d/Arduino.app/Contents/Java/libraries/WiFi
exec: "python": executable file not found in $PATH
Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

fmhotkeys, 2023-02-04:
Sketch uses 720398 bytes (36%) of program storage space. Maximum is 1966080 bytes.
Global variables use 39352 bytes (12%) of dynamic memory, leaving 288328 bytes for local variables. Maximum is 327680 bytes.
"C:\Users\GamerCircle\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1/esptool.exe" --chip esp32s2 --port "COM10" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 "C:\Users\GamerCircle\AppData\Local\Temp\arduino-sketch-F567FE0CF92BE76DEA6164640E2CCD7A/sketch_feb4a.ino.bootloader.bin" 0x8000 "C:\Users\GamerCircle\AppData\Local\Temp\arduino-sketch-F567FE0CF92BE76DEA6164640E2CCD7A/sketch_feb4a.ino.partitions.bin" 0xe000 "C:\Users\GamerCircle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\GamerCircle\AppData\Local\Temp\arduino-sketch-F567FE0CF92BE76DEA6164640E2CCD7A/sketch_feb4a.ino.bin" 
esptool.py v4.2.1
Serial port COM10
Connecting...
Chip is ESP32-S2
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: a0:76:4e:83:f4:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000bffff...
Compressed 14592 bytes to 10149...
Writing at 0x00001000... (20 %)
Writing at 0x00001c58... (40 %)
Writing at 0x0000279c... (60 %)
Writing at 0x000032ac... (80 %)
Writing at 0x00003c5b... (100 %)
Wrote 14592 bytes (10149 compressed) at 0x00001000 in 0.3 seconds (effective 391.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 320.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 448.5 kbit/s)...
Hash of data verified.
Compressed 720784 bytes to 488465...
Writing at 0x00010000... (0 %)
Writing at 0x00011355... (0 %)
Writing at 0x00011d2b... (1 %)
Writing at 0x0001251e... (1 %)
Writing at 0x00012d11... (2 %)
Writing at 0x000134ff... (2 %)
Writing at 0x00013cf1... (2 %)
Writing at 0x000144e0... (3 %)
Writing at 0x00014cd0... (3 %)
Writing at 0x00015486... (4 %)
Writing at 0x00015c7f... (4 %)
Writing at 0x0001647a... (5 %)
Writing at 0x00016c74... (5 %)
Writing at 0x0001746e... (5 %)
Writing at 0x00017c67... (6 %)
Writing at 0x00018461... (6 %)
Writing at 0x00018c5c... (7 %)
Writing at 0x00019845... (7 %)
Writing at 0x0001b075... (7 %)
Writing at 0x0001c71d... (8 %)
Writing at 0x0001db23... (8 %)
Writing at 0x0001f2e3... (9 %)
Writing at 0x00020d7a... (9 %)
Writing at 0x000225cd... (10 %)
Writing at 0x00024064... (10 %)
Writing at 0x00025420... (10 %)
Writing at 0x00026ca4... (11 %)
Writing at 0x0002758f... (11 %)
Writing at 0x00028648... (12 %)
Writing at 0x00029913... (12 %)
Writing at 0x0002aa41... (12 %)
Writing at 0x0002c893... (13 %)
Writing at 0x0002f356... (13 %)
[... 189 lines cut ...]
Writing at 0x000b4487... (93 %)
Writing at 0x000b4f79... (93 %)
Writing at 0x000b5acd... (94 %)
Writing at 0x000b66ae... (94 %)
Writing at 0x000b72ce... (94 %)
Writing at 0x000b7f6f... (95 %)
Writing at 0x000b8b52... (95 %)
Writing at 0x000b94ef... (96 %)
Writing at 0x000b9f48... (96 %)
Writing at 0x000ba9aa... (97 %)
Writing at 0x000bb394... (97 %)
Writing at 0x000bbe83... (97 %)
Writing at 0x000bca2f... (98 %)
Writing at 0x000bd794... (98 %)
Writing at 0x000be350... (99 %)
Writing at 0x000bed8b... (99 %)
Writing at 0x000bf974... (100 %)
Wrote 720784 bytes (488465 compressed) at 0x00010000 in 7.9 seconds (effective 730.0 kbit/s)...
Hash of data verified.

Leaving...
WARNING: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
Failed uploading: uploading error: exit status 1

LoneWarden93, 2023-04-21:
Yeah, Im having the same issue as fmhotkeys, stuck in download mode

yenul73, 2023-04-28:
Does anyone know how to get out of down load mode or how to reset the chip manualy cause holding boot then pressing reset on board does not fix!

bstory2001, 2023-05-26:
> Does anyone know how to get out of down load mode or how to reset the chip manualy cause holding boot then pressing reset on board does not fix!

Did you reset then check if the MauraderOTA wifi is broadcasting?  I also receive that error but was able to connect to the wifi and proceed with the update.  To be clear, I reset without holding boot.  The first time I tried it I made it past the login page but the device disconnected and stopped broadcasting on the page refresh step.  I reset the device and tried a second time and I was able to select the file and upload.  I think it isn't finished until the board flashes Ltblue, green, red regardless of the displayed progress%.