justcallmekoko/ESP32Marauder #227: Mount Error on external SD card

Darklisted, 2023-02-08:
I am using an ESP32-12K from waveshare flashed using the EZMarauderFlash for the FlipperZero I have connected an SD card breakout board (that has been tested and is working) with a FAT32 micro sd card (also working). The error is "Failed to mount SD Card SD Card NOT Supported".
 - Firmware version:  v0.10.1

Error occurs every time the device gets powered/rebooted

LOG CLI:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026515
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa74
load:0x40050000,len:0x25ec
entry 0x4004c194
ESP-IDF version is: v4.4.1-1-gb8050b365e
Failed to mount SD Card
SD Card NOT Supported
CLI Ready

              @@@@@@                        
              @@@@@@@@                      
              @@@@@@@@@@@                   
             @@@@@@  @@@@@@                 
          @@@@@@@      @@@@@@@              
        @@@@@@            @@@@@@            
     @@@@@@@                @@@@@@@         
   @@@@@@                      @@@@@@       
@@@@@@@              @@@@@@@@@@@@@@@@       
@@@@@                 @@@@@@@@@@@@@@@       
@@@@@                   @@@@@@@             
@@@@@                      @@@@@@           
@@@@@@                       @@@@@@@        
  @@@@@@                        @@@@@@@@@@@@
    @@@@@@@                          @@@@@@ 
       @@@@@@                     @@@@@@    
         @@@@@@@                @@@@@@      
            @@@@@@           @@@@@@         
              @@@@@@@      @@@@@@           
                 @@@@@@ @@@@@@              
                   @@@@@@@@@                
                      @@@@@@                
                        @@@@                
--------------------------------
         ESP32 Marauder      
            v0.10.1
       By: justcallmekoko
--------------------------------
```

Troubleshooting:
- Re-Program with the latest version
- Use of different firmware (Not the flipper zero version)
- Rewiring and testing ports 10, 9, and 11 for the CS of the breakout board

Darklisted, 2023-02-08:
Note that i have see a similar issue by  [sfernandez39](https://github.com/sfernandez39) [here](https://github.com/justcallmekoko/ESP32Marauder/issues/95) But it didn't help me

Kiwiers, 2023-02-21:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13472
load:0x40080400,len:3668
entry 0x400805f0
ESP-IDF version is: v4.4.1-1-gb8050b365e
Failed to mount SD Card
SD Card NOT Supported
CLI Ready
```
same here :/