justcallmekoko/ESP32Marauder #230: Devboard FlipperZero Wifi Network for OTA dont show up.

Visseb, 2023-02-12:
Describe the bug
I get to step 8 of [Installing Firmware via OTA]https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-via-ota) and the Wifi network "MaruderOTA" Never shows up, the board is still connectet to Arduino IDE, i followed the guide " https://www.youtube.com/watch?v=_YLTpNo5xa0"

Expected behavior
MarauderOTA dont show up.

Screenshots
n/a
Marauder (please complete the following information if applicable):

    Firmware version: n/a
    Hardware version: Flipperzero Devboard V1 brand new.

Desktop (please complete the following information if applicable):

    OS: Windows 11
    Browser: Chrome
    Version: n/a

Smartphone (please complete the following information if applicable):

    Device: n/a
    OS: n/a
    Browser n/a
    Version n/a

Additional context

napSec, 2023-02-12:
My issue isnt the same as your but it might be. I noticed that after the first reboot after the OTA upload it would show the marauder banner and if i rebooted it again it would just bring me back to the OTA upload page , when using the serial monitoring port

Visseb, 2023-02-23:
Anyone?

aki2273, 2023-04-16:
Same, did u resolve it?

Visseb, 2023-04-16:
> Same, did u resolve it?
> 

No :/

aki2273, 2023-04-16:
https://github.com/SkeletonMan03/FZEasyMarauderFlash

Do it with this link. Worked fine.

Visseb, 2023-04-16:
> https://github.com/SkeletonMan03/FZEasyMarauderFlash
> 
> Do it with this link. Worked fine.

im sorry but i didnt understand how to use the tools to install it with windows, i have downloaded all the files but its python scripts.

Visseb, 2023-04-17:
> https://github.com/SkeletonMan03/FZEasyMarauderFlash
> 
> Do it with this link. Worked fine.
https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher
I followed this guide.

kre8tivkulture, 2023-05-19:
Mine also didn't show up via pc. But it showed up on the phone wifi.What I did was connected it via android phone, downloaded the bin on my phone. I then connected to the up and ota'd it via android phone

Harris2shotKbke, 2023-06-02:
i have been having the same issue. finish the install and no network called marauder OTA shows up. i flashed multiple times using the fz marauder flasher and Arduino ide

m0veax, 2023-06-18:
I had the same Problem.

After changing Flash Size to "4MB (32MB)" and Partition Scheme to "Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)" it started working.

I am not sure if the wiki is outdated on that settings? I was able to connect to the "MarauderOTA" Wifi, Upload the binary afterwards and start using the Wifi Marauder Firmware on my Flipper Zero afterwards.

jorgedeand, 2023-08-25:
> I had the same Problem.
> 
> After changing Flash Size to "4MB (32MB)" and Partition Scheme to "Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)" it started working.
> 
> I am not sure if the wiki is outdated on that settings (I have fairly no knowledge about that topic)? I was able to connect to the "MarauderOTA" Wifi, Upload the binary afterwards and start using the Wifi Marauder Firmware on my Flipper Zero afterwards.

This worked for me. Thanks!