justcallmekoko/ESP32Marauder #234: MarauderOTA access point is constantly enabled

Gooman-rus, 2023-03-03:
**Describe the bug**
When I power on my flipperzero esp32 with marauder firmware I can see MarauderOTA in the list ov available access points. Is there any way to turn it off?

**Expected behavior**
MarauderOTA access point should turn on only after execution of the command `update -w`.

**Marauder (please complete the following information if applicable):**
 - Firmware version: MarauderOTA access point
 - Hardware version: FlipperZero board

MrJeems, 2023-03-20:
Are you able to connect to the Marauder software (either via serial or a Flipper Module) to successfully Scan APs?

I recently attempted to upgrade my Dev Board, but just before also happened to update my Arduino software and received board updates as well.
The OTA Flasher was written successfully, but the Marauder toolkit was only available from the time of Flash up until I removed power from the board.  As soon as I removed power it forgot the bin and reverted back to the OTA Flasher tool which enabled the access point and didn't give me access to the toolkit.
I was reading back through the documentation and it said to make sure the Board Manager was using v2.0.3 of "esp32 by Espressif Systems".  After my updates I was running v2.0.7.
I uninstalled the board from the IDE, reinstalled version 2.0.3, and re-compiled/flashed the DevBoard.
I uploaded the bin with the OTA utility and now Marauder functions as it should.

Gooman-rus, 2023-03-20:
There is no problem with functionality of marauder firmware itself. It works fine.

I recently also noticed that when I power on the eps32 board there is no MarauderOTA access point. But when I start deauth the MarauderOTA access point is activated.

justcallmekoko, 2023-03-20:
Fixed in v0.10.2

closed 2023-03-20 by justcallmekoko