justcallmekoko/ESP32Marauder #246: Error compiling marauder firmware for ttgo t-display to use with flipper

xxCLIMBINGMANxx, 2023-04-09:
Hey, i am trying to compile marauder firmware for an esp32 ttgo t-display because the default serial pins are not broken out so I am trying to define new ones. But I get this error when I compile on arduino IDE. Something about multiple definitions for a few things. 

Any help would be appreciated

Here is the error block:

`c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Buffer.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table4'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Buffer.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table3'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Buffer.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:184: multiple definition of `text_table2'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:184: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Buffer.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:183: multiple definition of `text_table1'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:183: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Buffer.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:182: multiple definition of `text_table0'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:182: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Display.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table4'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Display.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table3'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Display.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:184: multiple definition of `text_table2'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:184: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Display.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:183: multiple definition of `text_table1'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:183: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\Display.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:182: multiple definition of `text_table0'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:182: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\MenuFunctions.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table4'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\MenuFunctions.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table3'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\MenuFunctions.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:184: multiple definition of `text_table2'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:184: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\MenuFunctions.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:183: multiple definition of `text_table1'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:183: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\MenuFunctions.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:182: multiple definition of `text_table0'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:182: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\SDInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table4'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\SDInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table3'; sketch\BatteryInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\SDInterface.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Mara [... cut ...]
c:/users/cohen/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\cohen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; sketch\WiFiScan.cpp.o:C:\Users\cohen\OneDrive\Documents\ESP32Marauder-master\esp32_marauder/WiFiScan.cpp:13: first defined here
collect2.exe: error: ld returned 1 exit status
`

justcallmekoko, 2023-04-10:
https://github.com/justcallmekoko/ESP32Marauder/wiki/faq#multiple-definition-of-ieee80211_raw_frame_sanity_check

justcallmekoko, 2023-04-10:
Duplicate to #181

closed 2023-04-10 by justcallmekoko

xxCLIMBINGMANxx, 2023-04-10:
Thank you so much and sorry for the duplicate..