justcallmekoko/ESP32Marauder #249: Empty pcap with SD Serial support (Flipper Zero app) on ESP-WROOM-32

coded-with-claws, 2023-04-10:
Hi,
I compiled Marauder 0.10.2 branch develop (checked out yesterday) with WRITE_PACKETS_SERIAL and MARAUDER_V4, and uploaded it to my ESP-WROOM-32 (with Arduino IDE following the instructions from the Wiki), connected with double serial on my Flipper zero (Unleashed firmware with companion app supporting the SD serial pcap saving).
I say "double serial" because I got a first TX/RX couple for the Marauder communication (sending commands and reading results like AP list) and a second TX/RX couple for the pcap SD card saving.

When I run a "sniff pmkid" or a "sniff raw", I get lines displayed (`Received EAPOL:` or `RSSI ... Ch... BSSID`) but I get empty pcap files (0 bytes for pmkid_0.pcap and raw_0.pcap) on the SD card of Flipper zero.

I soldered the connections for the pcap SD serial saving like it is expected:
    - ESP32: RX2 -> FZ: 15 (TX pin)
    - ESP32: TX2 -> FZ: 16 (RX pin)
And I checked the electrical continuity with a multimeter.

Is the SD serial support supposed to work on ESP32 (not ESP32-S2)?

**To Reproduce**
Steps to reproduce the behavior:
1. On Flipper Zero, open the application "[ESP32] WiFi Marauder" (v0.3.3)
2. Run `Sniff PMKID on channel`
3. Lines are displayed:  `Received EAPOL:` 
4. Check the pmkid_0.pcap on the Flipper zero's SD card: 0 bytes

**Expected behavior**
The pcap files should not be empty because packets were captured, according to the Marauder log.

**Marauder:**
 - Firmware version:  0.10.2 branch develop (checked out yesterday) 
 - Hardware version: OG V4

**Additional context**
Except this problem for writing pcaps, the Flipper zero app works nice with Marauder on my ESP32.

Thanks :)

justcallmekoko, 2023-04-10:
Did you enable the SavePCAP setting?

coded-with-claws, 2023-04-10:
Yes, I have enabled this setting in the Flipper zero app.
Sorry I forgot to give this precision.

Also, I have read the same kind of issue on #141 but I'm up to date. But maybe it's somehow related?

JuriG66, 2023-04-10:
Hi, i have same issue :(

D0mpie, 2023-04-11:
Hello, I have the same problem on my brand new Flipper Zero Wi-Fi Module v1 with an ESP32-S2-WROVER module.

The new files are written onto the internal SD of the flipper but are 0 bytes. I use the latest ESP32 Marauder v0.10.2 installed with [FZEasyMarauderFlash](https://github.com/SkeletonMan03/FZEasyMarauderFlash) and the [Flipper-Xtreme)
](https://github.com/ClaraCrazy/Flipper-Xtreme) firmware for the Flipper.

coded-with-claws, 2023-04-11:
Ok I found something.

On ESP-WROOM-32, we have "serial0 aka TX0/RX0" and "serial2 aka TX2/RX2", but I don't see any TX1/RX1 printed on the chip. That's why I had soldered TX2 and RX2.
The Marauder code uses "Serial1" for the pcap data saving, and uses the library esp32 which maps the "Serial0/1/2" with the GPIO pin numbers, into [HardwareSerial.cpp](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp):

```
#ifndef RX1
#if CONFIG_IDF_TARGET_ESP32
#define RX1 9
[...]
#ifndef TX1
#if CONFIG_IDF_TARGET_ESP32
#define TX1 10
```

I also found this kind of information on this [link:](https://stackoverflow.com/questions/60094545/esp32-softwareserial-library#60094913)
```
Serial0: RX0 on GPIO3, TX0 on GPIO1
Serial1: RX1 on GPIO9, TX1 on GPIO10 (+CTS1 and RTS1)
Serial2: RX2 on GPIO16, TX2 on GPIO17 (+CTS2 and RTS2) 
```

Problem: On ESP-WROOM-32, I don't see GPIO 9 and 10 :(

So, I did a modification into the marauder code so that I now get the pcap files filled.

I replaced "Serial1" with "Serial2" into the following lines:
```
esp32_marauder/Buffer.cpp:      Serial1.write(bufB, bufSizeB);
esp32_marauder/Buffer.cpp:      Serial1.write(bufA, bufSizeA);
esp32_marauder/Buffer.cpp:      Serial1.write(bufA, bufSizeA);
esp32_marauder/Buffer.cpp:      Serial1.write(bufB, bufSizeB);
esp32_marauder/esp32_marauder.ino:    Serial1.begin(115200);
```
Now it works on my ESP-WROOM-32!

I'm not sure about the correct way to solve the problem:
- either modify Marauder code so that in OG V4 we use Serial2 for pcap data
- either re-solder on the good pins (but do they exist??)
- either modify HardwareSerial.cpp (I guess we should not do that!)

@justcallmekoko I guess you'll have a more enlightened opinion than me :)

Masimi, 2023-04-14:
I had the same problem with the Flipper Zero Wifi Dev Board, I search in the logs and I find a error menssage:
`
#sniffpmkid -c 33
Set channel: 33
Starting PMKID sniff on channel 33. Stop with stopscan
AP config set error, Maurauder SSID might visible : err=0x102
Received EAPOL:`

paolocamm, 2023-04-16:
Hi, i  have same problem with empty pcap files like you. ESP32 WROOM  i put the last update thanks  koko

codecolorado, 2023-04-17:
Same esp-wroom-32 on f0. Empty pcap files.  I have 100% faith we will see a fix soon.    Let me ask though. If I solder on an external card will it work?

clipboard1, 2023-04-18:
@coded-with-claws hi! i tried edit code like u said, but it does'nt works, can u said where did u connect rx2 tx2 ?

coded-with-claws, 2023-04-18:
Hi @clipboard1 I connected like I explained into the description:
- ESP32: RX2 -> FZ: 15 (TX pin)
- ESP32: TX2 -> FZ: 16 (RX pin)

clipboard1, 2023-04-19:
@coded-with-claws I did everything as you described, but I still get 1kb pmkid pcaps. What I did:
1.  uncomment WRITE_PACKETS_SERIAL and MARAUDER_V4
2. Replace Serial1 to Serial2 as you said
3. Turned on SavePCAP in marauder app on flipper 
Maybe i missed something...

coded-with-claws, 2023-04-19:
@clipboard1 
It seems you did everything like me...
I have an ESP-WROOM-32.
Can you test a "sniff raw" on a channel having wifi activity? (that's what I did)
Because people saw errors in logs about pmkid sniffing, and I still never managed to capture those, so I guess there is something to fix about that.

codecolorado, 2023-04-24:
> hi to all i just fix it   i deleded  marauder and i  installed esp32 penetretion tool  it works well 

Wait what?   I'm trying to get pcaps to save on flipper that surely wouldn't work

s3lfdstrukt, 2023-04-30:
I have this issue on esp32-s2. Empty PCAP, logs work fine.

codecolorado, 2023-05-05:
I'd rather not.   The entire point is to be able to properly use marauder.

s3lfdstrukt, 2023-05-09:
> you have to use `esp32_marauder_v0_10_3_20230414_flipper_sd_serial.bin` rather than `esp32_marauder_v0_10_3_20230414_flipper.bin`

I've tried both with the latest version. Now no logs or pcap.

codecolorado, 2023-05-09:
I believe they only work on the official dev board. I don't believe they work on the esp-wroom-32.

s3lfdstrukt, 2023-05-09:
> I believe they only work on the official dev board. I don't believe they work on the esp-wroom-32.

I have the official dev board

Codeined, 2023-05-10:
> Hi, i have same problem with empty pcap files like you. ESP32 WROOM

SAME HERE

D0mpie, 2023-05-14:
> Hello, I have the same problem on my brand new Flipper Zero Wi-Fi Module v1 with an ESP32-S2-WROVER module.
>
> The new files are written onto the internal SD of the flipper but are 0 bytes. I use the [...]

I just updated my Flipper to Xtreme firmware XFW-0046 06-05-2023. Cloned a new version of  FZEasyMaurauderFlash. After installing the requirements it downloaded esp32_marauder_v0_10_4_20230505_flipper_sd_serial.bin. Then I flashed the official Wifi dev board with option 2 (Flash SD Serial Marauder on Devboard or ESP32-S2). Now my pcap files aren't empty anymore. Not sure what fixed it. Nevertheless I wanted to share this. Hope some will find it helpful.

sbenne, 2023-05-15:
> Ok I found something.
>
> On ESP-WROOM-32, we have "serial0 aka TX0/RX0" and "serial2 aka TX2/RX2", but I don't see any TX1/RX1 printed on the chip. That's why I had soldered TX2 and RX2. The Marauder [...]

I didn't try, but after a lot of different flash, I think it's good start, I have take some time to understand which of the - "serial0 aka TX0/RX0" and "serial2 aka TX2/RX2" -, I have to use, just a correction it is not TX0/RX0 but TXD/RXD, like Dev/Debug ? if I am not wrong this could have sens, thanks

s3lfdstrukt, 2023-05-15:
> I just updated my Flipper to Xtreme firmware XFW-0046 06-05-2023. Cloned a new version of FZEasyMaurauderFlash. After installing the requirements it downloaded [...]

I tried this exact process and with the same hardware and I have no logs or pcaps at all. not empty ones, nothing.

sysfu, 2023-05-21:
I just updated the Wifi Dev Board firmware to 0.10.4 using Easy Flipper Zero Marauder Flash. https://github.com/SkeletonMan03/FZEasyMarauderFlash

Chose option 2 in the flash script as suggested by @chrisbward and now the pcaps are saving to the SDCard!

1337speed1337, 2023-05-24:
The following worked for me and the flipper wifi dev board v1 and extreme firmware (XFW-0046_06052023) but should work with unleashed as well:

Go here https://esp.huhn.me/

Connect to device and put the following bins in the appropriate place:

0x1000 = FZEasyMarauderFlash-main\Extra_ESP32_Bins\Marauder\bootloader.bin
0x8000 = FZEasyMarauderFlash-main\Extra_ESP32_Bins\Marauder\partitions.bin
0xE000 = empty
0x10000 = FZEasyMarauderFlash-main\ESP32Marauder\releases\esp32_marauder_v0_10_4_20230505_flipper_sd_serial.bin

Obviously powercycle and plugin and do your thing. Pcaps now saved to disk.

seanism, 2023-05-24:
Initially I selected 1 and had the zero bytes issue.  Reflashed with option 2 and now my pcap files are saving correctly.

Wifi dev board v1

codecolorado, 2023-05-24:
Listen. This thread was about the esp-wroom-32.    You guys have corrupted the thread with your flipper dev boards and what not...    I'm starting a new thread for the wroom.

ivanpfigueiredo, 2023-06-09:
> The following worked for me and the flipper wifi dev board v1 and extreme firmware (XFW-0046_06052023) but should work with unleashed as well:
>
> Go here https://esp.huhn.me/
>
> Connect to device and [...]

ty

f1yaw4y, 2023-06-09:
> 

Amazing work finding this solution. I guess another solution would be to solder directly to gpio 9 and 10 on the chip as oppossed to the breakout board, but then the expansion board would look ridiculous.

Which file did you modify to change these lines? I am also using ESP-WROOM-32

`esp32_marauder/Buffer.cpp:      Serial1.write(bufB, bufSizeB);
esp32_marauder/Buffer.cpp:      Serial1.write(bufA, bufSizeA);
esp32_marauder/Buffer.cpp:      Serial1.write(bufA, bufSizeA);
esp32_marauder/Buffer.cpp:      Serial1.write(bufB, bufSizeB);
esp32_marauder/esp32_marauder.ino:    Serial1.begin(115200);`

I soldered Tx0 and Rx0, it would be great to just change this to Serial0.begin, reflash and have it work. But I have no idea which firmware file you're referring to. Any help would be greatly appreciated.

Thank you much for the help and impressive job fixing the issue!

coded-with-claws, 2023-06-12:
Hi,
Thanks!
I changed the files directly into the cloned github project (https://github.com/justcallmekoko/ESP32Marauder/tree/master/esp32_marauder) and I compiled and flashed with Arduino IDE.

Shahar2k5, 2023-06-19:
> Hi, Thanks! I changed the files directly into the cloned github project (https://github.com/justcallmekoko/ESP32Marauder/tree/master/esp32_marauder) and I compiled and flashed with Arduino IDE. Tx0 and Rx0 should be reserved to the communication with the Flipper zero app (otherwise you will have to modify the app).

@coded-with-claws can you share the final result?
Which pins did you use? RX0/TX0 AND RX2/TX2? or just one pair?
Where did you connect them on the flipper?
Can you share the firmware bin file you compiled?

Thanks!

coded-with-claws, 2023-06-25:
@Shahar2k5 @caliKev 
I rebuilt the firmware for marauder 0.10.8 (interesting sniffpmkid feature on targeted AP :grin: ) with the same code patch I described earlier, and uploaded it for you on:
https://we.tl/t-Kp6TQw8zgp (expires in 1 week)

Like I said earlier, I had used the "sniff raw" feature to get pcap files, because I didn't manage to make "sniff pmkid" to work. But with the new developments on marauder, I hope to manage to do so!

Like I explained earlier, the serial wirings are:
- FZ 13/TX <=> ESP32 3/RX0
- FZ 14/RX <=> ESP32 1/TX0
- FZ 15/TX <=> ESP32 RX2
- FZ 16/RX <=> ESP32 TX2

@caliKev About you having corrupted pcap files, maybe you should check all you wirings with a multimeter. Make sure wanted connections, but also that unwanted connections are not present. But start checking the "sniff raw" first (wait for flipper displaying data so that you're sure you got packets).

Hope it helps you both make it work!

Balzermw, 2023-07-02:
> @Shahar2k5 @caliKev I rebuilt the firmware for marauder 0.10.8 (interesting sniffpmkid feature on targeted AP 😁 ) with the same code patch I described earlier, and uploaded it for you on: https://we.tl/t-Kp6TQw8zgp (expires in 1 week)
> 
> Like I said earlier, I had used the "sniff raw" feature to get pcap files, because I didn't manage to make "sniff pmkid" to work. But with the new developments on marauder, I hope to manage to do so!
> 
> Like I explained earlier, the serial wirings are:
> 
> * FZ 13/TX <=> ESP32 3/RX0
> * FZ 14/RX <=> ESP32 1/TX0
> * FZ 15/TX <=> ESP32 RX2
> * FZ 16/RX <=> ESP32 TX2
> 
> @caliKev About you having corrupted pcap files, maybe you should check all you wirings with a multimeter. Make sure wanted connections, but also that unwanted connections are not present. But start checking the "sniff raw" first (wait for flipper displaying data so that you're sure you got packets).
> 
> Hope it helps you both make it work!

can you please re-upload or send to me? on a ESP-WROOM-32 with the same issue

Hravid, 2023-07-03:
@Balzermw ur WeTransfer has already expired tho 😫😫😫

Asizon, 2023-07-03:
Hello all! Thank you very much for the solution @coded-with-claws. I cant make it work on my xiao c3 esp32 board because it only have 1 uart available, do you think that can be a solution on uart comunication to use two uart0 wires only?

coded-with-claws, 2023-07-03:
Here you go for a new upload (dunno the delay for this site) @Balzermw @Hravid 
https://www.udrop.com/KfDF/esp32_marauder.ino_(v0.10.8_ESP-WROOM-32_SD_serial).bin

@Shahar2k5 Great, happy to hear :)

@Asizon I don't know enough about ESPs to answer you, I guess you could search two other pins for the serial pcap communication and change the code into marauder's code, but it seems kinda complicated (to me at least!)

Asizon, 2023-07-04:
Looks like c3 also got an TX1/RX1 defined and available, where can i find serial1.write part code???

[image]

Asizon, 2023-07-05:
> Here you go for a new upload (dunno the delay for this site) @Balzermw @Hravid https://www.udrop.com/KfDF/esp32_marauder.ino_(v0.10.8_ESP-WROOM-32_SD_serial).bin
> 
> @Shahar2k5 Great, happy to hear :)
> 
> @Asizon I don't know enough about ESPs to answer you, I guess you could search two other pins for the serial pcap communication and change the code into marauder's code (and handle the pins with your own code, as they are not uart?!), but it seems kinda complicated (to me at least!)

I have managed to get another uart working on c3, i will post the solution later! I tried to make raw attack but i get 1kb file... Im not sure if it is working as expected

Asizon, 2023-07-09:
So here is the finnal easiest solution to configure uarts on wroom and xiao C·3
```
esp32 WROOM:
esp32_marauder/esp32_marauder.ino:    Serial1.begin(115200, 8N1, 15, 16);

Xiao C3:
esp32_marauder/esp32_marauder.ino:    Serial1.begin(115200, 8N1, 9, 10);
```
No need to modify buffer, just made this change.

coded-with-claws, 2023-07-09:
Hi @Asizon, great, thanks for the info!

Lorenzo-95, 2023-08-18:
> So here is the finnal easiest solution to configure uarts on wroom and xiao C·3
> 
> ```
> esp32 WROOM:
> esp32_marauder/esp32_marauder.ino:    Serial1.begin(115200, 8N1, 16, 17);
> 
> Xiao C3:
> esp32_marauder/esp32_marauder.ino:    Serial1.begin(115200, 8N1, 9, 10);
> ```
> 
> No need to modify buffer, just made this change.

Hi everyone,
I have the same issue with WROOM board and I wanna recap the solution you found. 

First of all, we have to edit esp32_marauder.ino line 195 like @Asizon commented, then compile with Arduino IDE. Then we can flash the compiled bin and solder the pins 16/17 with the Flipper pins C0/C1 (obviously inverting TX/RX). Is that right?

Should be this implemented in the next official release? @justcallmekoko

Lorenzo-95, 2023-08-21:
Hi,
I've tried to compile the firmware following step by step the documentation _Installing Firmware From Source_ but I have errors during the compiling process (all libraries installed correctly).
Has anyone the compiled firmware or a fix? I need serial on pins 16/17.
Thanks!

Lorenzo-95, 2023-10-04:
Hi, I'm still stuck :(
Anyone can help me please?

roripocket, 2023-10-17:
> I tried this exact process and with the same hardware and I have no logs or pcaps at all. not empty ones, nothing.

HI, I just notice that for ESP32-S2-WROVER the only firmare that write pcaps correctly to the internal SD is the next version : https://github.com/justcallmekoko/ESP32Marauder/releases/download/v0.10.8/esp32_marauder_v0_10_8_20230621_flipper_sd_serial.bin
Today is working fine.

LKloosterman, 2023-11-14:
> Hi, I compiled Marauder 0.10.2 branch develop (checked out yesterday) with WRITE_PACKETS_SERIAL and MARAUDER_V4, and uploaded it to my ESP-WROOM-32 (with Arduino IDE following the instructions from [...]

When you say "uploaded it to my ESP-WROOM-32" with the instructions, which Board Type did you select under Tools? For step 9 in the instructions:
"Select the appropriate ESP32 module under Tools>Boards

	Select LOLIN D32 for the following:
		Marauder v4 (OG)
		Marauder v6
		Marauder Mini
	Select Adafruit ESP32 Feather for the following:
		Marauder Kit
	Select ESP32-S2 dev board for the following:
		Marauder Flipper (Flipper Zero WiFi Dev Board)"

I'm unsure which to pick for a default/normal ESP-WROOM-32 board. LOLIN D32 since we're using Marauder v4 in the config.h? Or maybe just ESP32 DevKit?

coded-with-claws, 2023-11-19:
Hi,
I installed latest version (0.13.4), the SD write still works (ESP-WROOM-32).
But I still had to change the code. This time, I did like proposed by @asizon : 
`Serial1.begin(115200, SERIAL_8N1, 16, 17);` in `esp32_marauder.ino`
@justcallmekoko : would it be possible to add new "#defines" for WROOM like there is for XIAO_ESP32_S3 with XIAO_RX1 and XIAO_TX1?
Meaning:  
- into `configs.h` new lines for "ifdef ESP32_WROOM32" => `#define WROOM32_RX1 16` and `#define WROOM32_TX1 17`
- into `esp32_marauder.ino`, inside the `#ifdef WRITE_PACKETS_SERIAL`, new lines to test if we have ESP32_WROOM32, then run `Serial1.begin(115200, SERIAL_8N1, WROOM32_RX1, WROOM32_TX1);` 

I can try to work on it and push a merge request.

coded-with-claws, 2023-11-19:
> When you say "uploaded it to my ESP-WROOM-32" with the instructions, which Board Type did you select under Tools? For step 9 in the instructions: "Select the appropriate ESP32 module under [...]

Hi, I chose LOLIN D32 like advised in the wiki.

tmaiaroto, 2023-11-21:
@coded-with-claws do you have a fork of this repo somewhere with builds? I was trying to use act to run the git actions locally but builds failed. I think your adjustments are correct to support a larger number of boards.

coded-with-claws, 2023-11-21:
@tmaiaroto I pushed the changed on a fork and proposed a pull request: https://github.com/justcallmekoko/ESP32Marauder/pull/401
As for the 1 kb pcap files, I think it's not due to this present issue, but rather to the testing method. If you manage to get valid pcap files, your pin configuration is good.

tmaiaroto, 2023-11-22:
@coded-with-claws It did say they weren't valid pcap files. it wasn't like it stopped at 1kb. or maybe it did any something wasn't finished properly. but they wouldn't open in wireshark. i'm using pins 16, 17 on the esp-wroom-32 board marked D16 and D17 and put those into 15 and 16 on the FZ (right after the TX/RX). I'll try the bin files from your PR maybe I messed something up though all I did was change `Serial1.begin(115200);` to -> `Serial1.begin(115200, SERIAL_8N1, 16, 17);` in `esp32_marauder.ino`

coded-with-claws, 2023-11-22:
@tmaiaroto Do you test with "sniffraw"? (in order to be sure to get valid pcap files)
Maybe check the electric continuity of your wiring?

tmaiaroto, 2023-11-23:
@coded-with-claws I did use sniffraw. I'll see if I can somehow test the connections. I'm just using some dupont cables there, they're brand new. 🤷‍♂️ Something definitely populates the files I can leave it scan for some time and get larger files.

coded-with-claws, 2023-11-25:
I guess with brand new dupont cables it should be fine. You said you wired D16 and D17, maybe it's not the good pins?
On mine, they are named RX2 / TX2.
I have another board where there are no RX2 TX2 D16 D17 anywhere, so I guess there is no standard...
Do you have a picture of your board?

Asdii, 2023-11-26:
> @tmaiaroto I guess with brand new dupont cables it should be fine. You said you wired D16 and D17, maybe it's not the good pins? On mine, they are named RX2 / TX2. I have another board where there are no RX2 TX2 D16 D17 anywhere, so I guess there is no standard... Do you have a picture of your board?

I'm experiencing the same result. I used the .bin file you provided on a ESP-WROOM-32 (ESP32-D0WDQ6) from aliexpress, connected both RX/TX pins, and after sniffing raw data i get a file with some data in them, but i get the same error on Wireshark (commview: NCFX data length 0 < 20).

[image]

coded-with-claws, 2023-11-26:
@Asdii Your pins 16 & 17 are located exactly as my pins RX2 & TX2, and between I have the same pins... So your wiring seems good to me.
You're saying that you're using the .bin I provided, are you talking about when I uploaded it on wetransfer, or my github repository? Because it seems the .bin are not pushed, they still are dated of last year hence the date on the original justcallmekoko repository. On my disks they have the date of my build... But they are not listed on .gitignore so I don't understand why.
Here are the .bin I have built. As I don't know which one Arduino IDE uploads on the board, I provided the "d32.bin" and the "old_hardware.bin". The link will be up for one week.
https://we.tl/t-LNeDCfktNf
Please tell me if it's better?

tmaiaroto, 2023-11-26:
I'll give those a try, thanks? Before, I took the bin files from the GitHub actions that should have built the files pushed.

Asdii, 2023-11-26:
@coded-with-claws i used this link
https://www.udrop.com/KfDF/esp32_marauder.ino_(v0.10.8_ESP-WROOM-32_SD_serial).bin

I don't understand why it wouldn't work, i inspected some of the code inside buffer.cpp (I believe that's the one responsible of the serial writing) but couldn't find anything relevant, everything seems right. Maybe it's the flipper app that's expecting something different?, is there a way to see the Serial1 outpout before it's written to a file?

tmaiaroto, 2023-11-29:
I got a devboard from justcallmekoko. It doesn't seem to write to the flipper's sd card either, but it has an sd card slot of its own. I guess that's what most folks do. I'm more interested in solving this problem and making something custom than having a working unit. I don't exactly have any real use case here, it's more about learning and building something custom. So I guess I'll keep tinkering.

@coded-with-claws do you have a link to the board you bought? I'm curious if maybe I just got some flawed or incorrectly put togther ESP32-WROOM board. I only have one RX0/TX0 I don't see an RX2/TX2. Your fixes make complete sense to me based on what I've read. Switching to use different pins seems exactly right. Yet it doesn't seem to work on the board I have. I got some board off Amazon (a 3 pack they were so cheap) that has USB-C. The chip does not say expressif on it. It does say ESP-WROOM-32 though. So maybe I got a knock off or something that isn't implemented quite right.

Asdii, 2023-11-29:
I made this little test to see the output from the second serial communication by switching Serial1 with Serial 2 on the Flipper Zero and oppening Serial1 on PuTTy.

1.- Switch serial communications:
FZ RX -> ESP32 pin 17
FZ TX -> ESP32 pin 16

2.- Connect the ESP32 to your PC
3.- Open arduino IDE or PuTTy, open serial communication, scan ap's, select one and then sniffraw
4.- On the flipper zero select sniffraw (i know it doesn't work, but it shows communication anyway)

This is what i got:
[image]

@coded-with-claws could you do the same on your end to see if the data is simmilar? i'm trying to discard possible issues, and i know it's binary data, but at least it should be simmilar ascii.

Lorenzo-95, 2023-11-29:
> @Asdii Your pins 16 & 17 are located exactly as my pins RX2 & TX2, and in between I have the same pins... So your wiring seems good to me. You're saying that you're using the .bin I provided, are you talking about when I uploaded it on wetransfer, or my github repository? Because it seems the .bin are not pushed, they still are dated of last year hence the date of the original justcallmekoko repository. On my disks they have the date of my build... But they are not listed on .gitignore so I don't understand why they weren't pushed. So, here are the .bin I have built. As I don't know which one Arduino IDE uploads on the board, I provided the "d32.bin" and the "old_hardware.bin". The link will be up for one week. https://we.tl/t-LNeDCfktNf @Asdii @tmaiaroto Please tell me if it's better?

Hi, I tried flashing the compiled bin provided by @coded-with-claws on WeTransfer. Like @Asdii my pcaps are blank and the Wireshark error is the same, although pcaps dimension is not 0 anymore.

coded-with-claws, 2023-12-04:
> @coded-with-claws do you have a link to the board you bought? I'm curious if maybe I just got some flawed or incorrectly put togther ESP32-WROOM board. I only have one RX0/TX0 I don't see an [...]

It's this one I have: https://fr.aliexpress.com/item/1005001929935550.html?spm=a2g0o.order_detail.order_detail_item.5.11617d56iQx5cP&gatewayAdapt=glo2fra
=> click on the photo of "Color: 1pcs ESP-32".

At all: I don't have much time to help investigate, sorry.

[10 more replies]

closed 2024-02-13 by coded-with-claws