justcallmekoko/ESP32Marauder #261: Marauder on M5STICK C PLUS display half fuzzy

xxCLIMBINGMANxx, 2023-05-19:
Hey, 
I am trying to compile Marauder for the M5 Stick c Plus because I saw that someone has added compatibility for it. But it doesn't seem to be working. Marauder uploads fine and is functioning through the cli. But the display is not working properly. The left half of the display is displaying the right side of the picture and the right side of the display is just black and white fuzz. 
If anyone has any idea please let me know

Thanks

justcallmekoko, 2023-05-22:
This is because you need to set the offsets for the display within the TFT_eSPI library. Use my fork of the library

xxCLIMBINGMANxx, 2023-05-22:
I have your fork of the library installed but I can't find where to set the offsets. Can you let me know where to find them? 
Thanks

justcallmekoko, 2023-05-22:
You don't need to set the offsets then if you are using my fork. You just have to make sure you have the correct screen driver selected in the setup file for TFT_eSPI. You can use this one [here](https://github.com/justcallmekoko/ESP32Marauder/blob/develop/User_Setup_marauder_m5stickc.h)

xxCLIMBINGMANxx, 2023-05-22:
I’m am already using that config file but it still doesn’t display properly. I tried changing the offset values in the TFT driver folder under rotation but nothing happens.

justcallmekoko, 2023-05-22:
Could you provide a picture of the display while the firmware is running please

xxCLIMBINGMANxx, 2023-05-22:
Sure, sorry for the bad photo. The first image is after startup, showing the main menu. The second is during startup.

[image]
[image]

macgyverfuerarme, 2023-05-27:
I have the same issue...

xxCLIMBINGMANxx, 2023-05-28:
I have actually fixed the issue. If you want I can send you the binary of the updated firmware.

closed 2023-05-28 by xxCLIMBINGMANxx

alext30Developer, 2023-05-28:
> I have actually fixed the issue. The fixed firmware, uploader and instructions are listed here: https://discord.com/channels/776211399918878760/1108101475684581377

I can't view that. It says no text channels. Could you show it some other way?

xxCLIMBINGMANxx, 2023-05-29:
Not really. Github doesen't allow the .bin filetype. You probably haven't been joined the discord server properly

alext30Developer, 2023-05-29:
Would you mind sending a link to the discord server? Thanks!

macgyverfuerarme, 2023-06-09:
Even after the Update to 10.6 the bin file from here for the m5stickc plus has the screen issue. Only  a third of the screen is visible. 
With the 10.5 version from the discord server it works but the display in the menu is too squashed so that you cannot see the lower menu items when scrolling.

alext30Developer, 2023-06-26:
First join the owner's [discord server](https://discord.com/invite/Mn2d49Zaud)
Second follow the instructions [here](https://discord.com/channels/776211399918878760/1118383612073619497)
This worked for me!

Vimanik, 2023-10-12:
> hello! How can I install the esp32 marauder in m5stickc? I've tried several ways and I can't. thank you

Easiest way for beginner is to use UIFLOW FIRMWARE BURNING TOOL - https://docs.m5stack.com/en/download.