justcallmekoko/ESP32Marauder #267: Passive pmkid sniff deauth wifi clients

HanYangZhao, 2023-05-28:
**Describe the bug**
A clear and concise description of what the bug is.
Passive pmkid 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Assuming my wifi network is on CH1. When I run sniffpmkid -c 1, all clients will be disconnected from the AP and won't be able to reconnect until i stop running sniffpmkid. EAPOL are captured correctly

**Expected behavior**
A clear and concise description of what you expected to happen.
Since it's passive mode I'm expecting clients to not be disconnected from the AP

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.9.17] 10.0.5
 - Hardware version: [e.g. OG]  Flipper Zero with WIFI module

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information if applicable):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

joshmuente, 2023-06-02:
I can confirm.
the PKMID Sniff is disconnecting my Phone from Wifi.

HanYangZhao, 2023-06-03:
@joshmuente you're correct it does work as expected with ForcePMKID=disable. Is this an issue with the flipper app then?

joshmuente, 2023-06-04:
> @joshmuente you're correct it does work as expected with ForcePMKID=disable. Is this an issue with the flipper app then?

No not really. More like a documentation issue

closed 2023-06-25 by HanYangZhao

ALEEF02, 2023-08-15:
Can we get more documentation on ForcePMKID and ForceProbe?