justcallmekoko/ESP32Marauder #271: Reboot upon usb connect / disconnect

carnak, 2023-06-03:
**Describe the bug**

Anytime the usb cable between the marauder and the pc is connected / disconnected, the marauder device reboots.

**To Reproduce**
Steps to reproduce the behavior:
1. Simply connecting and/or disconnecting the usb cable between a pc and the marauder induces a reboot

**Expected behavior**
Don't constantly reboot

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: v0.10.4
 - Hardware version: v4.5 

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Putty
 - Version n/a

Distracto069, 2023-06-12:
Same happens to me

justcallmekoko, 2023-06-12:
This is a feature of the IP5306 charging chip and is not something that can be changed with the current hardware design.

closed 2023-06-12 by justcallmekoko