justcallmekoko/ESP32Marauder #273: Empty PCAP still

CyberGzuz, 2023-06-03:
Using flipper and latest xfw and the official flipper wifi dev board flashed with newest Marauder that just came out. I am still getting empty PCAPS.

justcallmekoko, 2023-06-03:
Can you post a screenshot of your empty pcaps?

CyberGzuz, 2023-06-03:
Let me scan-select-deauth-and scan raw- then save them and pull them. Then I’ll send pictures of what I am seeing.

CyberGzuz, 2023-06-03:
do you want a screenshot of the file properties or uploaded in wireshark?

CyberGzuz, 2023-06-03:
[image]

CyberGzuz, 2023-06-03:
[image]

justcallmekoko, 2023-06-03:
What is the version of Marauder you are running?
What is the hardware the Marauder firmware is running on?
Is this saving to the Flipper SD or the Marauder SD?

CyberGzuz, 2023-06-03:
10.5
on flipper zero wifi module v1 dev board from their site (no sd card)
saving to flipper sd
i used flipper.bin but i am currently trying again with serial.bin

CyberGzuz, 2023-06-03:
i now have a pcap file lol sorry for taking your time sir and thank you for the firmware!!!

closed 2023-06-03 by CyberGzuz