justcallmekoko/ESP32Marauder #275: Unresponsive CLI when trying to connect using macOS and flipper zero.

jean182, 2023-06-05:
***Is your feature request related to a problem? Please describe.**
I'm having trouble connecting my ESP32 with marauder installed using my flipper UART. bridge. I followed the CLI wiki https://github.com/justcallmekoko/ESP32Marauder/wiki/cli and even though I managed to connect the marauder terminal, it is very unresponsive and this makes it very hard to use.

*Describe the solution you'd like*
A section in the wiki on how to properly do this in a macOS device and I would love to understand why it is so slow and laggy.

*Describe alternatives you've considered*
I don't have any personal windows device and Putty is not available in macOS.

**Steps to repro**
1. Follow the [wiki](https://github.com/justcallmekoko/ESP32Marauder/wiki/cli) on how to enable UART(baudrate 115200) and connect the flipper to the computer
2. Get flipper name using this command
 `ls /dev/tty.*`
3. Copy the result and connect using this command:
 `screen /dev/tty.usbmodemflipFLIPPER_NAME 115200`
4. A connection with a blank screen is made, pressing the reset button in the ESP32 should launch the Marauder CLI
5. Typing is very laggy and it goes to the next line after a few milliseconds, so you have to type the command very fast in order to be able to run it.

[image]
As you can see when I input slow like in the li case it will go to the next line and it won't let me complete the command.

Salamafet, 2023-07-21:
In my case on MacOS I don't use `/dev/tty.usbmodemflip_FLIPERNAME1` I use `/dev/cu.usbmodemflip_FLIPPERNAME1`.

b1scoito, 2023-11-08:
Using this website helped me, it's way faster. https://serial.huhn.me/

JohnnyEn, 2024-05-01:
Hi @jean182 . I have discovered few great tools for serial connection.
GUI tool:
CoolTerm, neat standalone serial console with loads of custom config
https://freeware.the-meiers.org

CLI tool:
https://github.com/tio/tio
CLI console tool (afaik linux, maybe it's available for MAC too). You can enable line mode which is exactly what do you need. With this setting on it's possible to write commands in the CLI and send them by submit to the device. No more russian roulette with the blind typing.

You can naturally try to find any other tool, the key is find line mode of prompt submitting to the wifi dev board.