justcallmekoko/ESP32Marauder #281: M5Stickc plus Display issues

macgyverfuerarme, 2023-06-12:
Please update the .bin file for m5stickc plus. On your discord is a fixed bin for the 10.5 version. The bin files here are displaying only a third screen. I wonder which .bin you used in your YouTube short. The menu in this short is mush less smushed and looks more useful.

closed 2023-06-22 by macgyverfuerarme

f3shveg, 2023-06-26:
There are still some issues with screen after 0.10.8 update, i've seen a vertical orientation option in configs, i think this will help

mik3l333, 2023-06-28:
[macgyverfuerarme](https://github.com/macgyverfuerarme)[f3shveg](https://github.com/f3shveg)
Have you been able to install it? Could you explain to me how for dummies? thank you

@macgyverfuerarme
@f3shveg

f3shveg, 2023-06-28:
I think you using bin file from 0.10.8 release for M5StickC,  in that case you need to clone this git and open it as sketch, change everything you want to change in configs and then create bin file from sketch, I believe you can use same boot loader,partions you only need to change firmware to your new bin file
[Sketch to bin guide](https://randomnerdtutorials.com/bin-binary-files-sketch-arduino-ide/)

mik3l333, 2023-06-28:
> I think you using bin file from 0.10.8 release for M5StickC, in that case you need to clone this git and open it as sketch, change everything you want to change in configs and then create bin file [...]

66/70% of what you have told me sounds Chinese to me...
Could you explain it to me more simply? thank you.

f3shveg, 2023-06-28:
Ok, in a few hours

mik3l333, 2023-06-28:
> Ok, in a few hours

thank you so much my friend!

f3shveg, 2023-06-28:
Ok,I’ve ran through config and I haven’t seen vertical option and ngl I’ve ran into some problems while compiling, I think I’ll open new threat, so what we have now is the best we can get

laranja3, 2023-06-30:
Hello. 
I'm having the same problem.

I saw koko had some stuff here:
https://github.com/justcallmekoko/ESP32Marauder/commit/df613f3b9b8d4efc3671519075009bab2a6fbc39
I found it here:
https://github.com/justcallmekoko/ESP32Marauder/tree/df613f3b9b8d4efc3671519075009bab2a6fbc39

But idk if it helps...

I installed v0.10.8 here btw:
https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8

Do you know if i can just place the files from the 2 first links in https://esp.huhn.me/ and done?

Thank you!

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened to me before, when i first installed this firmware(the ESP32Marauder v0.10.8).However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then I deleted them from my device.
After it, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install the marauder firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
Thanks @macgyverfuerarme (;
But my device won't turn on!!!!!!!
Could someone explain to me how can i solve this?

laranja3, 2023-07-02:
it works now. I had to let the battery die.