justcallmekoko/ESP32Marauder #291: M5StickC Plus Freezing and No Screen after flashing .bin

TheTest21, 2023-06-28:
I am trying to get marauder working on my M5StickS Plus for hour and nothing is working I have tried using the ota but it shows this. 

```
exec: "cmd": executable file not found in %PATH%

Compilation error: exec: "cmd": executable file not found in %PATH%
```
and when I use the esptool, the M5 just freezes up and when trying to reboot it, the screen just doesn't even turn on.

Thanks in advance.

mk333333, 2023-06-28:
> You can use espwebtool, download boot loader,partions,boot app, and firmware and just flash them through site

But which ones do you have to download? because it doesn't put which ones are for the M5StickC PLUS!
ESP32 Marauder v4, v6, Kit, Mini?
Flipper Zero WiFi Dev Board
Flipper Zero Multi Board S3?
WiFi Dev Board Pro/LDDB/NodeMCU-32S/ESP32 Wemos D1 Mini?

From which of the 4 groups do you have to download the files?

f3shveg, 2023-06-28:
try this 
[ESP32Marauder](https://github.com/justcallmekoko/ESP32Marauder/tree/master)/[FlashFiles](https://github.com/justcallmekoko/ESP32Marauder/tree/master/FlashFiles)
/MarauderV4/

f3shveg, 2023-06-28:
#

> But which ones do you have to download? because it doesn't put which ones are for the M5StickC PLUS! ESP32 Marauder v4, v6, Kit, Mini? Flipper Zero WiFi Dev Board Flipper Zero Multi Board S3? WiFi [...]

So everything good ig?

TheTest21, 2023-06-29:
update well i tried the eswebtool thing and it didn't work but I got is working using the OTA and a raspberry pi 4 but the new issue is that when turned off apparently the .bin file disappears because when turned the screen is not even on.

f3shveg, 2023-06-29:
I can only give you a guide how to ude espwebtool
1.Burn your m5stickc (and go to [https://esp.huhn.me/](url))
2.Connect m5stick via usb and choose your port 
[image]
3.Then you will see this menu
[image]
4.Download all files from links and insert them in slots like on screenshot
[https://github.com/justcallmekoko/ESP32Marauder/raw/master/FlashFiles/MarauderV4/esp32_marauder.ino.bootloader.bin](url)
[https://github.com/justcallmekoko/ESP32Marauder/raw/master/FlashFiles/MarauderV4/esp32_marauder.ino.partitions.bin](url)
[https://github.com/justcallmekoko/ESP32Marauder/raw/master/FlashFiles/FlipperZeroMultiBoardS3/boot_app0.bin](url)
[https://github.com/justcallmekoko/ESP32Marauder/releases/download/v0.10.8/esp32_marauder_v0_10_8_20230621_m5stickc_plus.bin](url)
[image]
5.Click program
If you get errors like this 
[image]
I reccomend you to burn your m5stick again, i used m5burner for that

f3shveg, 2023-06-29:
[image]

TheTest21, 2023-06-29:
Can you explain how you used m5 burner I have tried to burn the .bin using it but I couldn't figure it out. And also did you use espwebtool with the arduino code?

f3shveg, 2023-06-29:
> Can you explain how you used m5 burner I have tried to burn the .bin using it but I couldn't figure it out. And also did you use espwebtool with the arduino code?

No, i only used bins that were uploaded by @justcallmekoko, i tried to compile it myself but i got 'firsy defined here' error and i gave up.
So, for the burner
1.Download the firmware(ui flow for example
[image]
2.Press burn(you can skip wifi configuration)
[image]
3.Press erase

TheTest21, 2023-06-29:
only thing that has worked for me has been using the OTA and Arduino but my m5 has to stay on or else everything erases.

mk333333, 2023-06-29:
> I can only give you a guide how to use espwebtool 1.Burn your m5stickc (and go to [https://esp.huhn.me/](url)) 2.Connect m5stick via usb and choose your port [image] 3.Then you will see this menu [image] 4.Download all files from links and insert them in slots like on screenshot [https://github.com/justcallmekoko/ESP32Marauder/raw/master/FlashFiles/MarauderV4/esp32_marauder.ino.bootloader.bin](url) [https://github.com/justcallmekoko/ESP32Marauder/raw/master/FlashFiles/MarauderV4/esp32_marauder.ino.partitions.bin](url) [https://github.com/justcallmekoko/ESP32Marauder/raw/master/FlashFiles/FlipperZeroMultiBoardS3/boot_app0.bin](url) [https://github.com/justcallmekoko/ESP32Marauder/releases/download/v0.10.8/esp32_marauder_v0_10_8_20230621_m5stickc_plus.bin](url) [image] 5.Click program If you get errors like this [image] I reccomend you to burn your m5stick again, i used m5burner for that

Following this guide I was able to install it.
The only thing I would add is that when the flash finished installing it would stay frozen without being able to do anything or go with any button. 
You have to unplug it and wait for the battery to run out. When it turns off and you plug it back in. 
it works now!!!!

THANK YOU SO MUCH!

f3shveg, 2023-06-29:
> Following this guide I was able to install it. The only thing I would add is that when the flash finished installing it would stay frozen without being able to do anything or go with any button. You [...]

You don’t have to wait that much,press and hold left button for like 7-8 seconds and then release and press it once

TheTest21, 2023-06-30:
I fixed it! I was using using the MaruderV4 boot ap instead of the flipper one and also got the cmd issue with these steps:

Check your system's PATH variable: Verify that the PATH environment variable is correctly set up on your computer. The PATH variable should include the directory where the "cmd" executable is located. By default, the "cmd" executable is typically found in the "C:\Windows\System32" directory. You can check the PATH variable by following these steps:

- Open the Start menu and search for "Environment Variables."

- Click on "Edit the system environment variables" to open the System Properties window.

- In the System Properties window, click on the "Environment Variables" button.

- Look for the "Path" variable under the "System variables" section and click on "Edit."

- Ensure that the path to the "C:\Windows\System32" directory is included in the list of paths. If it is missing, you can add it by clicking on "New" and entering the directory path.

by ChatGPT

closed 2023-06-30 by TheTest21

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened to me before, when i first installed this firmware(the ESP32Marauder v0.10.8).However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then I deleted them from my device.
After it, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install the marauder firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
But my device won't turn on!!!!!!!
Could someone explain to me how can i solve this?

laranja3, 2023-06-30:
Do i need the let the battery die?
I already tried to press 7-8s and 1s again but nothing!
@TheTest21 
@f3shveg 
@mk333333
Help!
#fordummies

laranja3, 2023-07-02:
it works now. I had to let the battery die.