justcallmekoko/ESP32Marauder #292: M5StickC Plus, impossible to install.

mik3l333, 2023-06-28:
I have tried all the ways I can think of to install the *.bin, but I have not succeeded.
Could someone explain to me how to install it on the M5StickC Plus for dummies, easily and step by step?
thank you so much

shanejohn15, 2023-06-29:
i have my m5stickc plus running marauder 10.8, ive tried so many ways on the m5stack core2 esp32, as well and i can not figure it out the arduino ide way, and im not sure if i was able to flash it through the web updater, or by them5burner because its now listed under m5stickc plus but when i originally flashed it it didnt show up, i waited a day and went to try it agian and it turned on with marauder on it. tbh i wish i knew which way i flashed it so i could you it on the bigger screen and more buttons on my m5stack 2. the drivers are listed in all the arduino files in the setup.h and config.h files but when i go to compile the sketch it comes back with countless errors with the code. ive tried all the different python/.bat files and they run and pull the correct esp32d0wdq6 v3. if someone can explain what to do for the m5stack with the esp32 in it please and thank you, or since its already written for the m5stickc maybe just change it over im not sure how m5ui works but if we can get it on there thatd be awsome. 
i really feel like these 2 devices have alot of potential closer to the flipper zero over anything else, if it was possible to get marauder on one of these as well as get the ir remote already made for the m5stickc into the marauder app, and keep databases on the sd card for the m5tack even get the little keyboard attachment for it and that would be a sick/fun little hacking device, especially since theres gpio pins all over it, can ovbiously be connected to the flipper but with an IR remote part of marauder and say something like spectrum analyzer with a cc1101 attached idk just thoughts but i wish i could be more help then just testing the hardware and firmware lol but ive had these m5tack and stickc for about a year now and just finally got something useful out of the m5stickc by getting Marauder working.

macgyverfuerarme, 2023-06-29:
I flashed mine this way:

Go on site: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8

Than Download the latest bin for the m5stickc plus.
Than click in Install/Update instructions.
Download the bootloader, partitions and boot App for the marauder v4 (first row).
Open the webupdater from spacehuhn.
Insert the four files and flash... Done.

laranja3, 2023-06-29:
Hello! I just flashed my m5stickc plus like you did  @(@macgyverfuerarme). However, i'm a potato and idk what to do next (,: I tried to turn it off and on but nothing... What do i need to do for it to run like in this video i just found: https://www.youtube.com/shorts/3Hb6KITZ8kg
Do i need to use arduino ide or uiflow?

My device won't turn on after with firmware...
Help! @shanejohn15 @macgyverfuerarme @mk333333 
#4dummies

laranja3, 2023-06-29:
Hello.
Do you know how to fix the screen?
I saw koko had some suff but I'm a potato🥔. Therefore, I need a tutorial. #4dummies
Thanks!

laranja3, 2023-06-29:
Hello.
Do you know how to fix the screen?
I saw koko had some stuff but I'm a potato🥔. Therefore, I need a tutorial. #4dummies
@macgyverfuerarme
Thanks!

laranja3, 2023-06-30:
Pls help me!

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened before, when i first installed this firmware. However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then deleted them from my device.
Then, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install this firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
Thanks @macgyverfuerarme (;
But my device won't turn on!
Could someone explain to me how can i solve this?

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened to me before, when i first installed this firmware(the ESP32Marauder v0.10.8).However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then deleted them from my device.
After it, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install this firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
Thanks @macgyverfuerarme (;
But my device won't turn on!
Could someone explain to me how can i solve this?

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened to me before, when i first installed this firmware(the ESP32Marauder v0.10.8).However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then I deleted them from my device.
After it, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install this firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
Thanks @macgyverfuerarme (;
But my device won't turn on!
Could someone explain to me how can i solve this?

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened to me before, when i first installed this firmware(the ESP32Marauder v0.10.8).However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then I deleted them from my device.
After it, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install the marauder firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
Thanks @macgyverfuerarme (;
But my device won't turn on!!!!!!!
Could someone explain to me how can i solve this?

laranja3, 2023-07-02:
it works now. I had to let the battery die.