justcallmekoko/ESP32Marauder #295: M5StickC Plus won't turn on

laranja3, 2023-06-30:
Hello.
My device won't turn on.
It has the right firmware.
It has full battery.
This has already happened to me before, when i first installed this firmware(the ESP32Marauder v0.10.8).However, after 30m it turned on and went well.
I tried some other stuff(other frimwares) for fun but then I deleted them from my device.
After it, i tried the M5burner with the title "M5StickCPlus-FactoryTest
official" and it ran smoothly.
Then, i deleted the firmware again.

NOW I'm trying to install the marauder firmware again (from here: https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.10.8)
Thanks @macgyverfuerarme (;
But my device won't turn on!!!!!!!
Could someone explain to me how can i solve this?

laranja3, 2023-06-30:
If you need, you can ask something related to the M5StickC Plus here as well. I will try to help 😁

laranja3, 2023-07-02:
it works now. I had to let the battery die.

laranja3, 2023-07-02:
How can i close this thing?

laranja3, 2023-07-02:
hmmmmm...

meekerton, 2023-07-15:
> If you need, you can ask something related to the M5StickC Plus here as well. I will try to help 😁

hello, did you just let the battery die, and then recharge it, and it suddenly ran marauder perfectly fine?

shanejohn15, 2023-07-20:
Have you looked into the rtlduino or rtl8720dn dev board? It's got 5ghz capabilities and I haven't heard anything from the dev I created an issues asking about it? I am just interested in having 5g marauder. This board uses its own libs called ambed in audrino. They were on Amazon for 3 for $23. I can't find any esp32 c5 which I believe yas 5g capabilities as well but these rtl8720dn also called bw16 dev board.

laranja3, 2023-07-26:
> > If you need, you can ask something related to the M5StickC Plus here as well. I will try to help 😁
> 
> hello, did you just let the battery die, and then recharge it, and it suddenly ran marauder perfectly fine?

Yes, the answer is yes. I apologize for taking so long to text back @meekerton!
If you want to ask something else feel free! (but i'm kinda noob with this stuff 😅)
If you have interesting news about marauder/M5Stick Plus i would really like to know!

laranja3, 2023-07-26:
> rtl8720dn dev board

No, but it looks REALLY exciting @shanejohn15!! 
I think they will not make a marauder 5GHz capable because it starts to be 
just a little illegal in some countries 👀 ... idk tbh...
Hey @meekerton! Do you know anything about this?
If you have any news on this or in any other 5GHz capable device with marauder let me know!
If you have other questions i will try to help!

shanejohn15, 2023-07-26:
@laranja3 I believe if it's all used for research purposes then it's not, if you use marauder as is now with just bt and 2.4g on a network or device that's not your own or you dont have permission on then it's illegal anyway.

But for pentesting reasons and research I believe alot of people are interested in 5ghz and the only other chip I've heard of was the esp32 c5 having 5ghz capabilities but that there like really really hard to get your hands on 

but the rtlduino is available now, so to me looks like the best option is rtl8720dn.

dacvvvide, 2023-07-27:
> Se hai bisogno, puoi anche chiedere qualcosa relativo a M5StickC Plus qui. Proverò ad aiutare😁

ciao ho il tuo stesso problema e dato che tu l'hai risolto volevo chiederti come fare perchè sono passati trenta minuti e comunque non si accende e non so cosa fare

laranja3, 2023-07-27:
> @laranja3 I believe if it's all used for research purposes then it's not, if you use marauder as is now with just bt and 2.4g on a network or device that's not your own or you dont have permission on then it's illegal anyway.
> 
> But for pentesting reasons and research I believe alot of people are interested in 5ghz and the only other chip I've heard of was the esp32 c5 having 5ghz capabilities but that there like really really hard to get your hands on
> 
> but the rtlduino is available now, so to me looks like the best option is rtl8720dn.
> 
> also tried to post an issue on @justcallmekoko Esp32marauder page for it, at least just wanted to know if he would be interested or if other people would be. I don't know how to code but I'm willing to help other ways, testing and however else

Good luck @shanejohn15 🤙