justcallmekoko/ESP32Marauder #299: Generic ESP with touch display as shown in the wiki does not work

Zobrob, 2023-07-09:
I have an ESP32 Wroom Nodemcu together with an ili9341 touch display. I hooked the pins up as described in the wiki.
What I tried:
1. I followed the guide and there was an error about a missing SD_CS definition while compiling. So I added that and tried again:
2. I uploaded the script with the modification without errors. Now I can connect with putty but the LCD is completely dark.
3. I tried to enable HAS_SCREEN but then I get an error in the library:
... \documents\arduino\libraries\lvgl\src\lv_conf_internal.h:41:18: fatal error: ../../lv_conf.h: No such file or directory
         #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
                  ^~~~~~~~~~~~~~~~~

LucaCraft89, 2023-08-09:
u have to comile as maraider v4 (OG ) by setting it in the configs.h

Zobrob, 2023-08-09:
Hi Luca, thanks a lot. I always tried with generic. Now I can see the menu but there are artifacts on the left side of the screen and touch is not mapped correctly. Any hints for this?

LucaCraft89, 2023-08-09:
No problem

Zobrob, 2023-08-18:
I just got one. Now I get the proper image but touch is automatically clicking the topmost item. I have to try if it works after I soldered the SD wires (I thought this is optional but maybe not).