justcallmekoko/ESP32Marauder #306: Flippers zero marauder evil portal not working );

ezpzX1, 2023-07-30:
**Describe the bug**
A clear and concise description of what the bug is.
Whenever you try to use evil portal on the Flipper zero marauder, even with the latest firmware version on the Wi-Fi module, it says ” could not find /ap.config.txt “ I have tried putting it in the apps, data folder, and in the  route and it is still not working. I’ve asked a bunch of my friends and they can’t figure out why it’s not working either. It might be a bug or it could just be me being dumb.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Evil portal to function put out a SSI ID and when you click on it, it takes you to a website asking for your Google Chrome username and password
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.9.17]
 - Hardware version: [e.g. OG] 

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information if applicable):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

eried, 2023-07-31:
> I have tried putting it in the apps, data folder

use [apps_data](https://github.com/bigbrodude6119/flipper-zero-evil-portal/tree/main#install-the-pre-built-app-on-the-flipper)

justcallmekoko, 2023-07-31:
Make sure you read through [this documentation](https://github.com/justcallmekoko/ESP32Marauder/wiki/evilportal) for how to use Evil Portal on Marauder.

mikensen66, 2023-08-01:
not working for me either. if i understand correctly - some wifi dev boards have separatly integrated slots for sd cards? is it not possible to load ap.config.txt and index.html from the SD card in the flipper zero device?

justcallmekoko, 2023-08-01:
Correct. In the documentation I sent, it states Evil Portal in Marauder requires there to be an SD card attached to the ESP32

ezpzX1, 2023-08-17:
> the

Why? Is it not possible to add it where it's in the flippers SD card? I don't feel like many people have the SD card mod installed. Thanks

closed 2023-08-26 by ezpzX1