justcallmekoko/ESP32Marauder #310: inverted touch vertically

LucaCraft89, 2023-08-09:
**Describe the bug**
the touch screen input is inverted vertically on the y axis

**To Reproduce**
Steps to reproduce the behavior:
1. use the draw app and find out

**Expected behavior**
working normally no inverted touch

**Marauder (please complete the following information if applicable):**
 - Firmware version: latest (0.11rc3)
 - Hardware version: DIY

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Browser: chromium based Microsoft Edge
 - Version 109

**Smartphone (please complete the following information if applicable):**
 - Device: Xiaomi Redmi Note 10 Pro
 - OS: MIUI 14.0.11
 - Browser chromium based Microsoft Edge
 - Version no idea

**Additional context**
i tried comiling for a lot of different hardware but same problem in the draw app and in the rest of the "OS" it is the same (along the y akis in the middle of the display the touch input doeas not match the content and i harlay have tried with multiple values of tft.rotation in the code but same issue)

LucaCraft89, 2023-08-09:
yes i arlady did do that bc i saw it but same issue but with new hardware is horizontaly mirrored

LucaCraft89, 2023-08-09:
Can u tell me where to put them? Bc i have already compiled from source

LucaCraft89, 2023-08-10:
Can u tell me where to put the values? because i have already compiled from source
also where is the example and how to use it

LucaCraft89, 2023-08-10:
@justcallmekoko

trisp3ar, 2023-08-11:
Wait a second, one time you say it's vertically inverted, then you say it's horizontally mirrored. Which one is it?!?
Also, what Marauder HW are you running?

LucaCraft89, 2023-08-11:
So it is vertically inverted if i compile for marauder v4 and horizontally inverted if i compile for v6 same story with the .bin the old_hardware is vertically inverted and the new_hardware.bin os horizontally inverted.

I am using an esp 32 wroom 32 and a tft touch screen display with the ili module for the sd card

trisp3ar, 2023-08-11:
That is valuable input. Since you are not using any official Marauder HW, I will not be able to help you out. Still, please provide a detailed list of the HW you are using for your build. We wanna keep this documented as best as possible.

LucaCraft89, 2023-08-11:
May I ask why u can't help me out?

And also i am using an esp32 wroom 32,  2.4" ILI9341 240x320 SPI TFT LCD Display 2.4 Inch Touch Panel LCD 5V, elegoo jumper cables and that is it

LucaCraft89, 2023-08-11:
@trisp3ar like @justcallmekoko was saying would i be alble to use che calibrate tft sketch? To resolve the issue?

LucaCraft89, 2023-08-12:
> That is valuable input. Since you are not using any official Marauder HW, I will not be able to help you out. Still, please provide a detailed list of the HW you are using for your build. We wanna keep this documented as best as possible.

If anything i am using the same hardware but they are not in the same PCB they are separate and i am not using a battery for the rest is the same exact esp32 and screen only difference that my touchscreen is physically smaller but the resolution is the same

LucaCraft89, 2023-08-16:
@justcallmekoko ?

justcallmekoko, 2023-10-10:
You can use the TFT_eSPI calibrate touch example sketch to get your calibration data. You would then use this calibration data to recompile the Marauder firmware specific to your display setup

LucaCraft89, 2024-07-15:
After long time figured it out i need to get the calibration data from TFT_eSPI calibrate example sketch and modify the source code to match

closed 2024-07-15 by LucaCraft89