justcallmekoko/ESP32Marauder #312: ESP32 NOT REBOOTING AFTER FIRMWARE UPDATE - mistake on bin file upload

KaminarimonOG, 2023-08-21:
**Describe the bug**
I uploaded via OTA the wrong bin file - the marauder did not reboot
after reboot the screen is black and the led are on
I am stuck
**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://192.168.4.1/
2. install latest version (I did not use OLD.HARDWARE.BIN by mistake
4. See error - not rebooting

**Expected behavior**
device reboot with new firmware updated
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v10.2 - >11]
 - Hardware version: [e.g. OG6] 

**Desktop (please complete the following information if applicable):**
 - OS: [windows 11]
 - Browser [brave 1.57.49]

**Smartphone (please complete the following information if applicable):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
I see other users experience the same issue but found no clear answer or reply

KingGhost12, 2023-10-08:
Yeah I did the same thing last night,bricked both of them I was figuring we have to rewrite the firmware from scratch...have you found a solution i asked a few places but to no avail 
[image]

justcallmekoko, 2023-10-10:
They are not bricked. You will need to use the spacehuhn web updater to reinstall the correct firmware

closed 2023-10-10 by justcallmekoko