justcallmekoko/ESP32Marauder #314: Is it possible to make evil portal not need the SD card mod?

ezpzX1, 2023-08-26:
I don't really understand why you didn't just make a folder for it in apps data. Is it possible to do that? If not could you tell me why? Thanks

FalcoGer, 2023-08-29:
Assuming you talk about the flipper dev board, as a workaround you can make a dual boot image for the esp32 with both the evilportal image and marauder.

You can then use the ESP Flasher application to switch between the two.

To do that you need to get the marauder firmware from the [release page](https://github.com/justcallmekoko/ESP32Marauder/releases) and the [release](https://github.com/bigbrodude6119/flipper-zero-evil-portal/releases) from evil portal.

Then you install esptool. assuming you are using linux

```bash
mkdir dualboot
virtualenv .
source bin/activate
python -m pip install -U esptool
```

After that you generate the image with

```
python -m esptool --chip esp32s2 merge_bin --target-offset 0x1000 0x1000 EvilPortal.ino.bootloader.bin 0x8000 EvilPortal.ino.partitions.bin 0xE000 boot_app0.bin 0x10000 EvilPortal.ino.bin 0x150000 esp32_marauder_v0_11_0_rc3_20230727_flipper_sd_serial.bin --output dual.bin
```
You then use the esp flasher application on flipper to flash the resulting dual.bin file to bootloader [0x1000].

You can use the esp flasher application to quickly switch between the two firmware options and then either start the marauder app (or use the usb-uart bridge) or the evil portal app (or again, use the usb-uart bridge)

justcallmekoko, 2023-10-10:
@ezpzX1 Since the communication between the ESP32 and the Flipper partially relies on the Marauder companion app, the companion app would need to be modified to function the same way as the Evil Portal companion app. As the companion app is not my project, this effort would need to be coordinated. For the time being, I've defaulted requiring an SD card to be attached to the ESP32.

closed 2023-10-10 by justcallmekoko