justcallmekoko/ESP32Marauder #322: 'class LinkedList<AccessPoint>' has no member named 'size'

DeLLyCeM, 2023-09-04:
Getting an error when compiling from source v0.11.0-RC3
'class LinkedList<AccessPoint>' has no member named 'size' from EvilPortal.cpp
LinkedList is latest v1.3.3

Any ideas ?

LouDnl, 2023-09-06:
Having the same issue here

LouDnl, 2023-09-06:
Could it be that ESPAsyncWebServer tries to use LinkedList defined in StringArray.h but ends up with LinkedList defined in LinkedList.h?

LouDnl, 2023-09-06:
It indeed seems that this is the case here, changing all LinkedList.h includes to StringArray.h (not a solution ofcourse) gets the compilation further but creates all sorts of other missing function calls obviously.

LouDnl, 2023-09-06:
Also mentioned in this issue https://github.com/me-no-dev/ESPAsyncWebServer/issues/1043

LouDnl, 2023-09-07:
> Getting an error when compiling from source v0.11.0-RC3
> 'class LinkedList<AccessPoint>' has no member named 'size' from EvilPortal.cpp
> LinkedList is latest v1.3.3
> 
> Any ideas ?

I can make a quick repo to pull this afternoon if you dont succeed.

DeLLyCeM, 2023-09-07:
Great tips, I am planning to use extracted ESPAsyncWebServer files from bigbrodude instead of Arduinos original library repo. Will try and find out when I have spare time. Thanks for the support and offer.

DeLLyCeM, 2023-09-07:
@LouDnl 

Solved
Main cause ESPAsyncWebServer from Arduino repo and updated esp32 boards which causes platform.txt file change. 

Solution;
Delete these libraries from Arduino directory; 

[lv_arduino]
[LinkedList]
[TFT_eSPI]
[JPEGDecoder library]
[NimBLE]
[NeoPixel]
[ArduinoJson]
[SwitchLib]
[ESPAsyncWebServer]
[AsyncTCP]

Download them from the link below and add them as .Zip library in Arduino>Sketch>Include Library
[https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-from-source](url)

Make sure you have esp32 boards version 2.0.10 installed in Arduino Boards Manager

Then edit platform.txt file as described in link below;
[https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup](url)

LouDnl, 2023-09-09:
> @LouDnl
>
> Solved
> Main cause ESPAsyncWebServer from Arduino repo and updated esp32 boards which causes platform.txt file change.
>
> Solution;
> Delete these libraries from Arduino directory; [...]

Good that you fixed it for yourself 💪🏻
Not exactly how I fixed it but it runs down to the same.

I have it running on an ESP32Cam with display, touch and sd. Not the built in sd but the one on the display.

wh1ter0z, 2023-09-26:
> Fixed it using the solution described in that issue. Ran into several other issues caused by the PROGMEM use in langvar not being used in the Arduino13+ way and that header being included in several places. Also EvilPortal has 2 variables that arent static. After these fixes everything finally compiles.
> 
> Now to get my esp32 back into flashing mode 🤣

can you help me ?
where need to fixed?

closed 2023-09-26 by justcallmekoko

sabas1080, 2023-11-21:
I have tested with the following library versions
LinkedList 1.3.3
ESPAsyncWebServer 1.2.3
AsyncTCP 1.1.1
ArduinoJson 6.19.4