justcallmekoko/ESP32Marauder #324: Menu items on the M5StickC Plus are outside the display area

SHUR1K-N, 2023-09-09:
**Describe the bug**
The menu items are shifted so far to the left on the M5StickC Plus's LCD screen, that I can only see "th" from "Bluetooth" (I think).

**To Reproduce**
Steps to reproduce the behavior:
1. Install v0.11.1 on the M5StickC Plus

**Expected behavior**
The menu items being on screen.

**Marauder (please complete the following information if applicable):**
 - Firmware version: 0.11.1

**Additional context**
The menu was fine on v0.11.0-RC3.

auburn5, 2023-09-19:
yes i have the same bug

Fiver007, 2023-09-28:
Yeah I'm having the same error. Last version that is working for me is 10.8

justcallmekoko, 2023-10-10:
https://github.com/justcallmekoko/ESP32Marauder/issues/341

closed 2023-10-10 by justcallmekoko