justcallmekoko/ESP32Marauder #340: Evil portal

Snifi0, 2023-09-23:
**Describe the bug**
The victim refreshes the page, marauder reboots.

**To Reproduce**
Steps to reproduce the behavior:
Coppy ap, launch evil portal, victim joins, refreshes page, boom. Can't take someone's login :<

**Expected behavior**
For it to refresh normally, not reboot

**Screenshots**
Na

**Marauder (please complete the following information if applicable):**
 OG

**Desktop (please complete the following information if applicable):**
Idk

**Smartphone (please complete the following information if applicable):**
Same

**Additional context**
Naa.

justcallmekoko, 2023-10-10:
What version of the Marauder firmware is this?

Snifi0, 2023-10-11:
@justcallmekoko 12