justcallmekoko/ESP32Marauder #341: marauder version 0.11.1 and 0.12.0 for M5StickCPlus the "background" is not centered like version 0.11.0-RC3

scheggia500, 2023-09-23:
I don't know why but in versions 0.11.1 and 0.12.0 the "background" is not centered so it is impossible to read the writing, but this problem does not occur in versions prior to 0.11.1 and 0.12.0. any help to see if this problem can be solved or will I have to wait for subsequent versions?
[image]

Fiver007, 2023-09-28:
Yeah I'm having the same error. Last version that is working for me is 10.8

PIckleRich22, 2023-10-17:
About to spend some time on this today but I am also having issues with the latest release. Last one properly functioning is 0.11.0-RC3

justcallmekoko, 2023-10-26:
Fixed https://github.com/justcallmekoko/ESP32Marauder/pull/367/commits/ea097190ac0380b3067c29ed6064775875bf7263
This will be implemented in v0.13.3

closed 2023-10-26 by justcallmekoko