justcallmekoko/ESP32Marauder #343: Compile Error

wh1ter0z, 2023-09-26:
https://filebin.net/jqm9ss8qcpkz6e8l

lvgl error
LinkedList error

Gregory88PL, 2023-09-27:
I have the same problems...
Do you resolve this?

wh1ter0z, 2023-10-04:
> I have the same problems... Do you resolve this?

Yes solved text me on Telegram @Thereisnomind

justcallmekoko, 2023-10-10:
What board are you compiling this for? What marauder hardware version do you have enabled in the config file? Need more context

MegaTheLEGEND, 2023-11-09:
```In file included from C:\Users\user\Downloads\ESP32Marauder-0.13.3\ESP32Marauder-0.13.3\esp32_marauder\Display.h:10,
                 from C:\Users\user\Downloads\ESP32Marauder-0.13.3\ESP32Marauder-0.13.3\esp32_marauder\Display.cpp:1:
C:\Users\user\Documents\Arduino\libraries\JPEGDecoder\src/JPEGDecoder.h:48:20: error: conflicting declaration 'fs::SPIFFSFS LittleFS'
     #define SPIFFS LittleFS
                    ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SPIFFS\src/SPIFFS.h:39:21: note: in expansion of macro 'SPIFFS'
 extern fs::SPIFFSFS SPIFFS;
                     ^~~~~~
In file included from C:\Users\user\Documents\Arduino\libraries\JPEGDecoder\src/JPEGDecoder.h:47,
                 from C:\Users\user\Downloads\ESP32Marauder-0.13.3\ESP32Marauder-0.13.3\esp32_marauder\Display.h:10,
                 from C:\Users\user\Downloads\ESP32Marauder-0.13.3\ESP32Marauder-0.13.3\esp32_marauder\Display.cpp:1:
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\LittleFS\src/LittleFS.h:39:23: note: previous declaration as 'fs::LittleFSFS LittleFS'
 extern fs::LittleFSFS LittleFS;
                       ^~~~~~~~
Multiple libraries were found for "SD.h"
  Used: C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SD
  Not used: C:\Users\user\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: exit status 1
```

possibly the same problem

wh1ter0z, 2023-11-09:
> ```
>                  from C:\Users\user\Downloads\ESP32Marauder-0.13.3\ESP32Marauder-0.13.3\esp32_marauder\Display.cpp:1: [...]

fixed text me in telegram i will help you

wh1ter0z, 2023-11-09:
> What board are you compiling this for? What marauder hardware version do you have enabled in the config file? Need more context

fixed sir

MegaTheLEGEND, 2023-11-09:
cant you answer here?

wh1ter0z, 2023-11-09:
need to give you libraries

wh1ter0z, 2023-11-09:
> cant you answer here?

u have to update custom libs

MegaTheLEGEND, 2023-11-09:
ok

wh1ter0z, 2023-11-09:
> ok

https://github.com/wh1ter0z/ESP32Marauder/tree/master/esp32_marauder libraries / libraries.rar 

customise your display in tft folder

MegaTheLEGEND, 2023-11-09:
Thx I will check it out

zapaidos, 2024-05-12:
I just solved this issue in a very simple way, go to the JPEGDecoder.h at the libraries forder open it, search for #define SPIFFS LittleFS and rename SPIFFS to SPIFFS_JPG or anything different from SPIFFS because SPIFFS is used at the display.h and we cannot have two declarations with the same name. Important this declaration is present twice in the JPEGDecoder.h so you need to change both