justcallmekoko/ESP32Marauder #345: When i done Flashing my M5stickc Plus the image Show Like this why ??  ESP32Marauder

Salomonas, 2023-10-02:
I tryed Twice and even Older Firmware 0.11 But Dident work what should i do 
i use M5stickC PLUS 

and the Picture of the Software Show It Out of the screen WHat should i do ? 

picture of the problem:   https://cdn.discordapp.com/attachments/701291217241243750/1158437928293367828/f8e8887d-dbf4-4566-8bb1-0fea8cbc4dbf.jpg?ex=651c3ee4&is=651aed64&hm=4c70cf06b63a2b329cbdb91042e7de0ed77d1714538ef0cfa1bad392c636447d&

Salomonas, 2023-10-02:
So after researching i found that v0.10.8 is working Thx for Noting :)

justcallmekoko, 2023-10-10:
The TFT_eSPI library was updated and has since changed the offset of the screen. I need to fix the the offset setting