justcallmekoko/ESP32Marauder #346: Wigle logging structure

zRCrackiiN, 2023-10-06:
When uploading a wardive long to wigle.net you'll find that it doesn't recognise any of the data in the file.

So I manually changed a small wardive log to test if I could get it to accept it and it works!

The changes needed are as follows
-Pre header
-header
-adjust date/time stamp, if the date is below 1 digit it must have a zero at the start eg. 4th of the month would be "04"

This explains the wigle API format in detail
https://api.wigle.net/csvFormat.html

Here's an example log, that is confirmed working and no, it won't dox my location, it's just a place I've driven through.

[WigleWifi-1.4],appRelease=[x],model=[ESP32S2],release=[x],device=[Flipper-dev],display=[MonochromeLCD],board=[FlipperDevBoard],Brand=[Flipper]
MAC,SSID,AuthMode,FirstSeen,Channel,RSSI,CurrentLatitude,CurrentLongitude,AltitudeMeters,AccuracyMeters,Type
A2:B5:3C:25:3C:3C,Telstra253C34,[WPA2_WPA3_PSK],2023-10-04 21:36:11,5,-77,-37.8144302,144.7577515,45.30,5.25,WIFI
F4:8C:EB:EE:1D:D2,HJLGUEST,[WPA2_PSK],2023-10-04 21:36:11,5,-82,-37.8144302,144.7577515,45.30,5.25,WIFI
14:59:C0:ED:56:B1,Telstra1B7D,[WPA2_PSK],2023-10-04 21:36:12,6,-58,-37.8144302,144.7577209,45.00,5.25,WIFI
EC:AD:E0:54:0A:D0,HJL RADIUS,[WPA2],2023-10-04 21:36:17,11,-81,-37.8144112,144.7577057,47.70,5.25,WIFI
EC:AD:E0:54:0A:D2,,[WPA2_PSK],2023-10-04 21:36:17,11,-82,-37.8144112,144.7577057,47.70,5.25,WIFI
EC:AD:E0:54:0A:D4,HJLWHPROD,[WPA2_PSK],2023-10-04 21:36:30,11,-82,-37.8144379,144.7576294,39.70,4.75,WIFI

justcallmekoko, 2023-10-16:
What did the log look like before you altered it?

zRCrackiiN, 2023-10-18:
```
#wardrive
Starting Wardrive. Stop with stopscan
> 1 | 2A:3F:0B:BB:D4:5A,Aries Rail,[WPA2_PSK],2023-10-4 21:36:7,1,-52,-37.8144379,144.7577515,46.30,4.75,WIFI
2 | A2:B5:3C:25:3C:3C,Telstra253C34,[WPA2_WPA3_PSK],2023-10-4 21:36:11,5,-77,-37.8144302,144.7577515,45.30,5.25,WIFI
3 | F4:8C:EB:EE:1D:D2,HJLGUEST,[WPA2_PSK],2023-10-4 21:36:11,5,-82,-37.8144302,144.7577515,45.30,5.25,WIFI
4 | 14:59:C0:ED:56:B1,Telstra1B7D,[WPA2_PSK],2023-10-4 21:36:12,6,-58,-37.8144302,144.7577209,45.00,5.25,WIFI
5 | EC:AD:E0:54:0A:D0,HJL RADIUS,[WPA2],2023-10-4 21:36:17,11,-81,-37.8144112,144.7577057,47.70,5.25,WIFI
6 | EC:AD:E0:54:0A:D2,,[WPA2_PSK],2023-10-4 21:36:17,11,-82,-37.8144112,144.7577057,47.70,5.25,WIFI
7 | EC:AD:E0:54:0A:D4,HJLWHPROD,[WPA2_PSK],2023-10-4 21:36:30,11,-82,-37.8144379,144.7576294,39.70,4.75,WIFI
#stopscan
```

DerSeb90, 2024-01-17:
Implement an ready to upload file to wigle would be awesome.

frankea, 2024-01-27:
Same issue here, debating coming up with a script to convert, unless someone else already has one?

frankea, 2024-01-28:
Here is my initial attempt at a conversion script which worked for me on one log, with some limitations as I haven't finished it out yet: https://github.com/frankea/stuff/blob/main/convert_wardrive.py

Limitations:
- Hard coded for WiFi at the moment.
- Doesn't convert timestamps to include leading 0 if missing (only works for logs after 12PM).
- May or may not work with multiple 'Capabilities'.  Not sure yet as I haven't gotten a log with that to test with (nor have I created one).

DerSeb90, 2024-11-06:
Any Updates on this? Is this fixed in Firmware v.1.0.0?

closed 2025-09-14 by github-actions[bot]