justcallmekoko/ESP32Marauder #347: pcap files are broken at  "_flipper_sd_serial" firmware

RekordNadoev, 2023-10-07:
Hardware:
Flipper Zero dev-board (ESP32-S2) with *_flipper_sd_serial.bin firmware

Software:
Flipper Zero Marauder companion v0.6.2
Saving .pcap files to Flipper's SD-card.

Saved .pcap files are broken after "esp32_marauder_v0_10_8_20230621_flipper_sd_serial.bin" firmware.

Open with tcpdump v4.99.1: "tcpdump: unknown file format"
Open with wireshark v3.6.2: "The file "sniffpmkid_0.pcap" appears to be damaged or corrupt (commview: NCFX data negth 0 < 20)"
Open with aircrack-ng v1.7: "Unsupported file format (not a pcap or IVs file)."

Tried firmwares:
esp32_marauder_v0_10_8_20230621_flipper_sd_serial.bin - OK
esp32_marauder_v0_11_0_rc3_20230727_flipper_sd_serial.bin - .pcap is broken
esp32_marauder_v0_11_1_20230907_flipper_sd_serial.bin - .pcap is broken
esp32_marauder_v0_12_0_20230910_flipper_sd_serial.bin  - .pcap is broken
esp32_marauder_v0_12_1_20231005_flipper_sd_serial.bin  - .pcap is broken

**To Reproduce**
Steps to reproduce the behavior:
1. Get Flipper Zero with dev-board + ESP32 Marauder firmware
2. Run Apps => GPIO => [ESP32] WiFi Marauder
3. Set "Save to flipper sdcard" to "Yes" for logs and pcaps
4. Run Sniff => pmkid (or Sniff => raw) => get .pcap files saved to flipper's sdcard
5. Copy .pcap files to PC and try to open with wireshark/tcpdump/aircrack-ng
6. Get error messages. tcpdump: "tcpdump: unknown file format"; wireshark: "The file "sniffpmkid_0.pcap" appears to be damaged or corrupt (commview: NCFX data negth 0 < 20)"; aircrack-ng: "Unsupported file format (not a pcap or IVs file)."

**Expected behavior**
Steps 1-5 are the same.
6. .pcap files opening OK with wireshark/tcpdump/aircrack-ng.

**Marauder (please complete the following information if applicable):**
 - Firmware version: 0_11_0_rc3 and later
 - Hardware version: Flipper Zero dev-board (ESP32-S2)

**Additional context**
Last tried version with correct .pcap files was esp32_marauder_v0_10_8_20230621_flipper_sd_serial.bin

thinkjk, 2023-10-10:
I'm having the same issues, I've tried a few fixes from reddit and none of worked.

justcallmekoko, 2023-10-10:
Have you tried this with the non serial SD version of the firmware and saving directly to an SD card connected to the ESP32?

RekordNadoev, 2023-10-10:
No unfortunately.
I've original FlipperZero dev-board with no SD only.
All i can tell: esp32_marauder_v0_10_8_20230621_flipper_sd_serial.bin does work, newer versions does not.

RekordNadoev, 2023-10-10:
Found ESP32 Marauder v6 hardware with SD and checked:
- esp32_marauder_v0_11_1_20230907_new_hardware.bin
- esp32_marauder_v0_12_1_20231005_new_hardware.bin
Both seemed to work correctly. aircrack-ng opened both of the .pcap dumps.

So the problem is in *flipper_sd_serial* after v0_10_8

What additional info should i provide to help debug?

thinkjk, 2023-10-10:
> ESP32 Marauder v6

I'm happy to help test on my dev board as well. and as @RekordNadoev mentioned there is no SD slot on the board by default.

justcallmekoko, 2023-10-14:
I am currently working on a solution. The issue first appears in v0.11.0 and I've found it's cause. This will be fixed in v0.13.1

RekordNadoev, 2023-10-14:
Thank you for great job!

justcallmekoko, 2023-10-14:
All set
https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.13.1

closed 2023-10-14 by justcallmekoko

thinkjk, 2023-10-15:
> All set https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.13.1

Confirmed it's working for me now. Thanks @justcallmekoko!

Ralhazrolex, 2024-01-11:
updated to v0.13.1 and still pcaps are zero bytes, what could I be missing?

eaudaim, 2024-05-25:
Hi ! 
Same here, updated to version 0.13.10 with fzeeflasher, and still have 0 bytes files.

Hardware is a flipper zero and esp-wroom-32.

When the marauder find the good channel (even on the "targeted active" mode, it failed to aim the selected acces point almost everytime), the flipper display this text : 
[image]

[image]

But the files "sniffpmkid_XXX.txt stay empty... :/

PS : Wiring is like this : 

> Esp32 RXD => TX Flipper zero,
> 
> ESP32 TXD => RX,
> 
> Vin ESP32 pin to 3.3v Flipper zero pin
> 
> GND Esp32 pin to GND Flipper zero pin_

RekordNadoev, 2024-05-27:
Tried ESP32-wroom-32 with esp32_marauder_v0_13_10_20240425_esp32_lddb firmware and Marauder companion v0.71.
It just worked.

Did You turn on "Save to flipper sdcard" in the app?

eaudaim, 2024-05-28:
Yes sir !

eaudaim, 2024-05-29:
Can you explain me in detail the procedure you followed to make it work ?
Did you flashed with FzeeFlasher or another tool like ArduinoIDE or something else ? 

Did you wired it the same way i did ?

RekordNadoev, 2024-06-06:
Yes, wired it the same way.

I use unleashed firmware dev branch.

- connect ESP32-wroom-32 to Flipper
- run Menu => Apps => GPIO => ESP Flasher
- at the ESP32-wroom-32 board: push "BOOT button", push "EN button", release "EN button", release "BOOT button"
- in "ESP Flasher" app slect "Quick Flash" => "ESP32-WROOM"

- run Menu => Apps => GPIO => [ESP32] WiFi Marauder
- select "Save to flipper sdcard" => yes => yes
- run any sniffing mode and get non-empty pcap in /ext/apps_data/marauder/pcaps

RekordNadoev, 2024-06-06:
Missed one step in my previous message.

After "in "ESP Flasher" app slect "Quick Flash" => "ESP32-WROOM"" (bootloader)
- in "ESP Flasher" app slect "Quick Flash" => "other ESP32-WROOM" => "Marauder (has EvilPortal)"

eaudaim, 2024-11-04:
I forgot to try your aswer for almost 5 month, but today i came back to this problem and your solution worked perfectly.

I noticed that with other techniques (fzeeflasher for example) marauder had various problems, and especially it did not stop once the sniff was launched, even when going back in the menus, and it created several PCAP files, I think the problem came from corrupted PCAP files because I had to reboot the board to stop the sniff.

What I find strange is that the first time I used the pinball application to flash the esp, some problems were there, others not, but I don't remember if the files were empty.

Anyway, thank you very much for your help :)