justcallmekoko/ESP32Marauder #354: M5StickC plus screen

lowskii, 2023-10-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.9.17]
 - Hardware version: [e.g. OG] 

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information if applicable):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

bro please fix the screen offset on m5stickcplus, also sour apple does not work

justcallmekoko, 2023-10-14:
Need some more context and descriptive language here. "Does not work" doesn't tell me much. How does it not work? What did you do to test?

I can verify on my end that it *does* work.

Also...duplicate https://github.com/justcallmekoko/ESP32Marauder/issues/341

closed 2023-10-14 by justcallmekoko

lowskii, 2023-10-14:
nothing happens when I launch this attack, my iPhone feels fine)