justcallmekoko/ESP32Marauder #359: apple spam on m5stick c plus

auburn5, 2023-10-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

auburn5, 2023-10-18:
Is there any way that you can make it so apple spam from flipper zero can run on M5 stick C plus

VinP55, 2023-11-12:
is there anyway you could make a easy manual/guidebook on functionality having trouble on couple items