justcallmekoko/ESP32Marauder #362: No work

samukas81, 2023-10-20:
Swiftpair Spam no work, tested in xiaomi poco f3 and dell Inspiron.
V4 old

justcallmekoko, 2023-10-20:
Need more info

samukas81, 2023-10-20:
Nothing simply happens, I believe it does not generate any data flow for the spam to work.

justcallmekoko, 2023-10-20:
What hardware are you using to run Marauder firmware?
If it's not official Marauder hardware, what type of ESP32 module are you using?
What version of Marauder firmware are you using?
If you are using a Flipper Zero, what are you using to interface with the Marauder firmware?
If you are using the Marauder companion app on the Flipper Zero, what version is it?
Do you have Swift Pair enabled on your target device? If so, share a screenshot of the checkbox

samukas81, 2023-10-20:
Esp32 devkit 1
firmware v4 old
no flipper
other functions work perfectly!

justcallmekoko, 2023-10-20:
What is the firmware version number and which bin file did you use? _old_hardware? _new_hardware? _lddb?
Please provide a picture of your devkit

closed 2023-11-15 by justcallmekoko