justcallmekoko/ESP32Marauder #369: ESP32-WROOM wifi marauder not fully working on Flipper0

casimirextreme, 2023-10-26:
**Describe the bug**
I'm using the latest Roguemaster firmware (RM1024-0158-0.93.1-1129e6e) and flashed the ESP32-WROOM (Chip is ESP32-D0WD-V3 (revision v3.1)) with FZEasyMarauderFlash with release is 0.13.3

Wifi Marauder doesn't provide good pcaps.
When starting a scan, a sniff or whatever action, I only got:
`
Received EAPOL:
Received EAPOL:
Received EAPOL:
..
`
There are no BSSID showed. The pcap is really small (but never empty) and shows as corrupted on Wireshark
On the other end, the Evil Portal is working fine.

**To Reproduce**
Start whatever action on GPIO/ESP32/WiFi Marauder app on F0 

**Expected behavior**
Should show the BSSID when receiving EAPOL and create good pcaps

laurent-luyckx-keemotion, 2023-11-29:
> same problem, had to compile a modified version of the esp32_marauder.ino corresponding on your own esp32 version

Can you share what should be modified? Thanks.

casimirextreme, 2023-12-06:
Closing as solutions found on https://github.com/justcallmekoko/ESP32Marauder/issues/249

Dowan35, 2023-12-07:
> Closing as solutions found on #249

Hello could you send me your esp32_marauder folder you used please ? :) I have so many errors when I try compiling myselft, thanks !

casimirextreme, 2023-12-07:
> > Closing as solutions found on #249
> 
> Hello could you send me your esp32_marauder folder you used please ? :) I have so many errors when I try compiling myselft, thanks !

I was also struggling to find the correct lib versions. I finally used those from @wh1ter0z (Kuddos if you see this)
https://github.com/wh1ter0z/ESP32Marauder/blob/master/esp32_marauder/libraries/libraries.rar

wh1ter0z, 2023-12-08:
> > > > Closing as solutions found on #249
> > > 
> > > 
> > > Hello could you send me your esp32_marauder folder you used please ? :) I have so many errors when I try compiling myselft, thanks !
> > 
> > I was also struggling to find the correct lib versions. I finally used those from @wh1ter0z (Kuddos if you see this) https://github.com/wh1ter0z/ESP32Marauder/blob/master/esp32_marauder/libraries/libraries.rar
> 
> Thanks for your answer, I installed the libs and used the esp32_marauder folder from wh1ter0z but I still have the 'WiFiScan.cpp:1573:5: error: 'esp_base_mac_addr_set' was not declared in this scope; did you mean 'esp_base_mac_addr'?'
>  when trying to compile on arduino uno, do you still have your esp32_marauder folder ? thank you !

What version of arduino ide u use ?

Dowan35, 2023-12-08:
> What version of arduino ide u use ?

I tried compiling with 2 versions: 1.8.18 and 2.2.1
on both versions I have the quoted error from the WifiScan.ccp file

wh1ter0z, 2023-12-08:
> > > > > > > > Closing as solutions found on #249
> > > > > > > 
> > > > > > > 
> > > > > > > Hello could you send me your esp32_marauder folder you used please ? :) I have so many errors when I try compiling myselft, thanks !
> > > > > > 
> > > > > > 
> > > > > > I was also struggling to find the correct lib versions. I finally used those from @wh1ter0z (Kuddos if you see this) https://github.com/wh1ter0z/ESP32Marauder/blob/master/esp32_marauder/libraries/libraries.rar
> > > > > 
> > > > > 
> > > > > Thanks for your answer, I installed the libs and used the esp32_marauder folder from wh1ter0z but I still have the 'WiFiScan.cpp:1573:5: error: 'esp_base_mac_addr_set' was not declared in this scope; did you mean 'esp_base_mac_addr'?'
> > > > > when trying to compile on arduino uno, do you still have your esp32_marauder folder ? thank you !
> > > > 
> > > > 
> > > > What version of arduino ide u use ?
> > > 
> > > 
> > > I tried compiling with 2 versions: 1.8.18 and 2.2.1
> > > on both versions I have the quoted error from the WifiScan.ccp file
> > 
> > You have to use 1.8 version not 2 sir
> > 
> > Use 1.8
> > 
> > And share photo of your error
> 
> Okay, just installed the 1.8.0 version instead of the 1.8.18 I installed. I imported your librairies and clicked upload and now I have a new error:
> [image]
> 
> 

Did u edit platform.txt ? 
In appdata/arduino15

Dowan35, 2023-12-08:
> > > > > > > > > Closing as solutions found on #249
> > > > > > > > 
> > > > > > > > 
> > > > > > > > Hello could you send me your esp32_marauder folder you used please ? :) I have so many errors when I try compiling myselft, thanks !
> > > > > > > 
> > > > > > > 
> > > > > > > I was also struggling to find the correct lib versions. I finally used those from @wh1ter0z (Kuddos if you see this) https://github.com/wh1ter0z/ESP32Marauder/blob/master/esp32_marauder/libraries/libraries.rar
> > > > > > 
> > > > > > 
> > > > > > Thanks for your answer, I installed the libs and used the esp32_marauder folder from wh1ter0z but I still have the 'WiFiScan.cpp:1573:5: error: 'esp_base_mac_addr_set' was not declared in this scope; did you mean 'esp_base_mac_addr'?'
> > > > > > when trying to compile on arduino uno, do you still have your esp32_marauder folder ? thank you !
> > > > > 
> > > > > 
> > > > > What version of arduino ide u use ?
> > > > 
> > > > 
> > > > I tried compiling with 2 versions: 1.8.18 and 2.2.1
> > > > on both versions I have the quoted error from the WifiScan.ccp file
> > > 
> > > 
> > > You have to use 1.8 version not 2 sir
> > > Use 1.8
> > > And share photo of your error
> > 
> > 
> > Okay, just installed the 1.8.0 version instead of the 1.8.18 I installed. I imported your librairies and clicked upload and now I have a new error:
> > ![Capture d’écran 2023-12-08 153401](https://private-user-images.githubusercontent.com/94186633/289100895-e6189919-6ad5-4782-99e4-8ae5e515c95c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwNDcyMzcsIm5iZiI6MTcwMjA0NjkzNywicGF0aCI6Ii85NDE4NjYzMy8yODkxMDA4OTUtZTYxODk5MTktNmFkNS00NzgyLTk5ZTQtOGFlNWU1MTVjOTVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA4VDE0NDg1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyNGY5N2Q2YWJkMTNlMjRkOWM0YTczNDdiNGU4YzRhODlmMzlhNzc2N2Y1ODlkM2NhNjIyZTUwZTMzYzMyZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bJGnXINxhwIqRkCPLkww-v6PqQ3OqNO-qt9T0TtbLHk)
> 
> Did u edit platform.txt ? In appdata/arduino15

Okay, I just did that, now I have one last error : 
[image]

thanks a lot for your help !

wh1ter0z, 2023-12-08:
> > > > > > > > > > Closing as solutions found on #249
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > Hello could you send me your esp32_marauder folder you used please ? :) I have so many errors when I try compiling myselft, thanks !
> > > > > > > > 
> > > > > > > > 
> > > > > > > > I was also struggling to find the correct lib versions. I finally used those from @wh1ter0z (Kuddos if you see this) https://github.com/wh1ter0z/ESP32Marauder/blob/master/esp32_marauder/libraries/libraries.rar
> > > > > > > 
> > > > > > > 
> > > > > > > Thanks for your answer, I installed the libs and used the esp32_marauder folder from wh1ter0z but I still have the 'WiFiScan.cpp:1573:5: error: 'esp_base_mac_addr_set' was not declared in this scope; did you mean 'esp_base_mac_addr'?'
> > > > > > > when trying to compile on arduino uno, do you still have your esp32_marauder folder ? thank you !
> > > > > > 
> > > > > > 
> > > > > > What version of arduino ide u use ?
> > > > > 
> > > > > 
> > > > > I tried compiling with 2 versions: 1.8.18 and 2.2.1
> > > > > on both versions I have the quoted error from the WifiScan.ccp file
> > > > 
> > > > 
> > > > You have to use 1.8 version not 2 sir
> > > > Use 1.8
> > > > And share photo of your error
> > > 
> > > 
> > > Okay, just installed the 1.8.0 version instead of the 1.8.18 I installed. I imported your librairies and clicked upload and now I have a new error:
> > > ![Capture d’écran 2023-12-08 153401](https://private-user-images.githubusercontent.com/94186633/289100895-e6189919-6ad5-4782-99e4-8ae5e515c95c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwNDcyMzcsIm5iZiI6MTcwMjA0NjkzNywicGF0aCI6Ii85NDE4NjYzMy8yODkxMDA4OTUtZTYxODk5MTktNmFkNS00NzgyLTk5ZTQtOGFlNWU1MTVjOTVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA4VDE0NDg1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyNGY5N2Q2YWJkMTNlMjRkOWM0YTczNDdiNGU4YzRhODlmMzlhNzc2N2Y1ODlkM2NhNjIyZTUwZTMzYzMyZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bJGnXINxhwIqRkCPLkww-v6PqQ3OqNO-qt9T0TtbLHk)
> > 
> > Did u edit platform.txt ? In appdata/arduino15
> 
> Okay, I just did that, now I have one last error : 
> [image]
> 
> 
> thanks a lot for your help !
> 
> (I did all the steps in https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-from-source , plus the
> Add -w to the following compiler settings
> build.extra_flags.esp32
> build.extra_flags.esp32s2
> build.extra_flags.esp32s3
> build.extra_flags.esp32c3
> Add -zmuldefs to the following compiler settings
> compiler.c.elf.libs.esp32
> compiler.c.elf.libs.esp32s2
> compiler.c.elf.libs.esp32s3
> compiler.c.elf.libs.esp32c3 )

Show me your file / line that u got error

Dowan35, 2023-12-08:
> 
> Show me your file / line that u got error
> 
> Also : "Adafruit_NeoPixel" install this lib

Adafruit was already installed : 
[image]

here is the full error:

WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
c:/users/gulli/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\LedInterface.cpp.o:(.literal._ZN12LedInterface8RunSetupEv+0x0): undefined reference to `strip'

collect2.exe: error: ld returned 1 exit status

I choose board ESP32 Dev Module in minimal spiffs and changed to GENERIC_ESP32 in configs.h

wh1ter0z, 2023-12-08:
> >  @wh1ter0z 
> > Show me your file / line that u got error
> > 
> > Also : "Adafruit_NeoPixel" install this lib
> 
> Adafruit was already installed : 
> [image]
> 
> here is the full error:
> 
> WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
> WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
> WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
> c:/users/gulli/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\LedInterface.cpp.o:(.literal._ZN12LedInterface8RunSetupEv+0x0): undefined reference to `strip'
> 
> collect2.exe: error: ld returned 1 exit status
> 
> I choose board ESP32 Dev Module in minimal spiffs and changed to GENERIC_ESP32 in configs.h

Hmmm 

That error u got it mean strip variable isnt defined ... 
But in LEDinterface is defined 

If u want to just using cli version you can edit code and remove strip variable 

Or if you want to use Gui with tft and you want led status you can fix it with defined a strip variable 

So if you want for flipper u can remove ... no need this

Dowan35, 2023-12-08:
> > > @wh1ter0z
> > > Show me your file / line that u got error
> > > Also : "Adafruit_NeoPixel" install this lib
> > 
> > 
> > Adafruit was already installed :
> > ![adafruit](https://private-user-images.githubusercontent.com/94186633/289124257-cabee0d7-4961-4108-b4cd-6c6fa83b1e53.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwNTI0MTcsIm5iZiI6MTcwMjA1MjExNywicGF0aCI6Ii85NDE4NjYzMy8yODkxMjQyNTctY2FiZWUwZDctNDk2MS00MTA4LWI0Y2QtNmM2ZmE4M2IxZTUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA4VDE2MTUxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhZDQ3NmNjNTM2MzA4ZTA2OGNiZWFhYjc0ZjMyOGMyNmJjYWQ4MjRjNTkzMDJiMGI4NTY1MTIxMDA2NzEyNDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4lzLDkbDc_X-_VX705L1QikdppCn2TopNcdbVufjwjs)
> > here is the full error:
> > WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
> > WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
> > WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
> > c:/users/gulli/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\LedInterface.cpp.o:(.literal._ZN12LedInterface8RunSetupEv+0x0): undefined reference to `strip'
> > collect2.exe: error: ld returned 1 exit status
> > I choose board ESP32 Dev Module in minimal spiffs and changed to GENERIC_ESP32 in configs.h
> 
> Hmmm
> 
> That error u got it mean strip variable isnt defined ... But in LEDinterface is defined
> 
> If u want to just using cli version you can edit code and remove strip variable
> 
> Or if you want to use Gui with tft and you want led status you can fix it with defined a strip variable
> 
> So if you want for flipper u can remove ... no need this
> 
> If u want fix it go RunSetup in LedInterface.cpp
> 
> And defined a variable or in .h file check if strip is private move it to public

I just commented all the lines talking about strip so it creates new errors, how could I define strip to avoid additional errors ? in effective I use the flipper so I dont need a GUI display but all is interconnected so it's difficult to just comment one line

wh1ter0z, 2023-12-08:
> > > > @wh1ter0z
> > > > Show me your file / line that u got error
> > > > Also : "Adafruit_NeoPixel" install this lib
> > > 
> > > 
> > > Adafruit was already installed :
> > > ![adafruit](https://private-user-images.githubusercontent.com/94186633/289124257-cabee0d7-4961-4108-b4cd-6c6fa83b1e53.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwNTI0MTcsIm5iZiI6MTcwMjA1MjExNywicGF0aCI6Ii85NDE4NjYzMy8yODkxMjQyNTctY2FiZWUwZDctNDk2MS00MTA4LWI0Y2QtNmM2ZmE4M2IxZTUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA4VDE2MTUxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhZDQ3NmNjNTM2MzA4ZTA2OGNiZWFhYjc0ZjMyOGMyNmJjYWQ4MjRjNTkzMDJiMGI4NTY1MTIxMDA2NzEyNDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4lzLDkbDc_X-_VX705L1QikdppCn2TopNcdbVufjwjs)
> > > here is the full error:
> > > WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
> > > WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
> > > WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
> > > c:/users/gulli/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\LedInterface.cpp.o:(.literal._ZN12LedInterface8RunSetupEv+0x0): undefined reference to `strip'
> > > collect2.exe: error: ld returned 1 exit status
> > > I choose board ESP32 Dev Module in minimal spiffs and changed to GENERIC_ESP32 in configs.h
> > 
> > Hmmm
> > 
> > That error u got it mean strip variable isnt defined ... But in LEDinterface is defined
> > 
> > If u want to just using cli version you can edit code and remove strip variable
> > 
> > Or if you want to use Gui with tft and you want led status you can fix it with defined a strip variable
> > 
> > So if you want for flipper u can remove ... no need this
> > 
> > If u want fix it go RunSetup in LedInterface.cpp
> > 
> > And defined a variable or in .h file check if strip is private move it to public
> 
> I just commented all the lines talking about strip so it creates new errors, how could I define strip in RunSetup of LedInterface.cpp to avoid additional errors ? in effective I use the flipper so I dont need a GUI display but all is interconnected so it's difficult to just comment one line

My ledinterface.cpp is ok 

Without error 
You can comment all variables include strip 

Or u can add this into cpp file of ledinterface 

Add "Adafruit_NeoPixel strip;" into Ledinterface.cpp 
Delete "extern Adafruit_NeoPixel strip;" in Ledinterface.h

Dowan35, 2023-12-08:
> > > @wh1ter0z
> > > Thank you seems to work with those changes ! I just dont have eapol in my pcap file on my flipper sd card but I will try to fix that
> 
> Your error fixed? With Adafruit_NeoPixel strip; ?

yep I can compile without errors with your fix, just have a little problem with the content of my pcap files that I will try to solve

Dowan35, 2023-12-08:
> @Dowan35 Have a look at the diff attached: [ESP32Marauder-0.13.5-WROOM.diff.zip](https://github.com/justcallmekoko/ESP32Marauder/files/13619029/ESP32Marauder-0.13.5-WROOM.diff.zip) This is what I did to have pcaps on my esp32-wroom

yup I did your changes, I added define HAS_SD in configs.h , my sd card has the pcap file, not empty but the file seems invalid when I try to convert it on cap2hashcat, will try with another access point

Dowan35, 2023-12-08:
> I don't think HAS_SD is necessary. Did you add?
> 
> ```
> //// SD DEFINITIONS
>   #ifdef GENERIC_ESP32
>     #define SD_CS 5
>   #endif
> ```

yup I did all of your changes ( last version from justcallmekoko with the sd change for the evil portal changes some things in the code also)

wh1ter0z, 2023-12-09:
> > > I don't think HAS_SD is necessary. Did you add?
> > > ```
> > > //// SD DEFINITIONS
> > >   #ifdef GENERIC_ESP32
> > >     #define SD_CS 5
> > >   #endif
> > > ```
> > 
> > It need
> > 
> > If you want use SdCard module
> > 
> > You have to define spi(V,H) If VSPI mosi ,miso , clk need to be (23,19,18) ans cs can be 5
> > 
> > Also You have to define what type of spi you wamt to use
> > 
> > #define SD_CS 5 #define SD_MOSI 23 #define SD_MISO 19 #define SD_CLK 18 #define SD_SPI_FREQ 10000000 SPIClass spiSD = SPIClass(VSPI);
> > 
> > In setup
> > 
> > spiSD.begin(SD_CLK, SD_MISO, SD_MOSI, SD_CS); if (!SD.begin(SD_CS, spiSD, SD_SPI_FREQ)) { Serial.println(F("Failed to mount SD Card")); this->supported = false; return false; }
> 
> Indeed. That is if you want to use an external sdcard. In my case, I just want to use the internal flipper zero card.
> Thanks.

Projects for flipper all are with external sdcard