justcallmekoko/ESP32Marauder #371: M5StickC Plus LED always on (v0.13.3)

SHUR1K-N, 2023-10-27:
**Describe the bug**
The red LED stays on regardless of the "EnableLED" setting's state. This impacts the internal battery's juice.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn the device on. The LED turns on and then never turns off during operation until the device is shut down

**Marauder (please complete the following information if applicable):**
 - Firmware version: v0.13.3

PS: Great update, by the way!

n0xa, 2023-10-29:
It's been that way for me since I first tried (0.10.8 release).

I'll look into it. It's getting on my nerves, too.

n0xa, 2023-10-29:
The M5StickC Plus powers the red LED and IR LED through the AXP192, and curiously, these LEDs are Active Low, that means when GPIO10 is LOW, current flows through the LEDs. A kind of strange design decision. You can see what I'm talking about in sector C3 of the schematic: https://static-cdn.m5stack.com/resource/docs/products/core/m5stickc_plus/m5stickc_plus_sch_03.webp

I'll tackle the LED on the M5Stick C Plus the same way The Flipper Marauder and Xiao are supported. I'm working on this now.

justcallmekoko, 2023-11-06:
https://github.com/justcallmekoko/ESP32Marauder/commit/21b5c4fcc65c6e13c963638383a86edd53f61691

closed 2023-11-06 by justcallmekoko