justcallmekoko/ESP32Marauder #379: Google Fast Pair and Kitchen Sink(randomize all) feature

Mirubzi, 2023-11-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Hello, my english is not very good, but i think you will understand me. I wan't to see google fast pair spam and kitchen sink(randomize all, apple, microsoft and android) feature in the future versions of esp32 marauder.
**Describe the solution you'd like**
I will try to found how to do it, but I'm not so strong at coding. 
**Additional context**
here is the service names of devices https://github.com/Flipper-XFW/Xtreme-Firmware/blob/dev/applications/external/ble_spam/protocols/fastpair.c
and another one site with instruction how to do it using android phone and custom messages https://www.mobile-hacker.com/2023/10/17/spam-ios-android-and-windows-with-bluetooth-pairing-messages-using-flipper-zero-or-android-smartphone/

Mirubzi, 2023-11-05:
here is the all files about ble spam https://github.com/Flipper-XFW/Xtreme-Firmware/tree/dev/applications/external/ble_spam

wh1ter0z, 2023-11-09:
use your phone for this options : https://github.com/simondankelmann/Bluetooth-LE-Spam/releases/tag/v1.0.2

wh1ter0z, 2023-11-13:
i updated and now have all this option you said

Mirubzi, 2023-11-13:
thank you very much

justcallmekoko, 2023-11-15:
@wh1ter0z Is this something you plan to add into a PR for Marauder?

wh1ter0z, 2023-11-16:
> @wh1ter0z Is this something you plan to add into a PR for Marauder?

I added { samsung buds spam , samsung watch spam, Edited Swift pair names , google fast pair spam (all content) } 

Also added random mode for Ble spam

Edited libs for poeples have problem for compile 

I have idea for ios < 15 / maybe it works for new ios 

Hexway apple research :
https://github.com/hexway/apple_bleee

[image]

[image]

wh1ter0z, 2023-11-16:
> Wow, it's really nice. I will check it when I will be have time

[image]

[image]

Mirubzi, 2023-11-16:
when system will be ready for tests?

wh1ter0z, 2023-11-16:
> ahhahahahaahahhah
> when system will be ready for tests?
> 

First i have to fix crash issue 
After that i will give source to Auther

Mirubzi, 2023-11-16:
> > ahhahahahaahahhah
> > when system will be ready for tests?
> 
> First i have to fix crash issue After that i will give source to Auther

ok. i'm waiting

wh1ter0z, 2023-11-16:
@justcallmekoko can you help me to fix it ?

ReZak7, 2024-03-12:
Hi, can you give me your project file?