justcallmekoko/ESP32Marauder #381: crash on swiftpair

wh1ter0z, 2023-11-09:
after about 3sec will crash , i changed delay from 10 to 100ms now about 10sec 
error with esp exception decode:

    Decoding stack result 
                0x4011cdd6: NimBLEDevice::host_task(void*) at C:\Users\rx\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp line 836

any help to fix it ?

coded-with-claws, 2023-11-19:
Hi,
I seem to encounter the same symptom, on ESP32-WROOM-32, but I haven't tested changing the delay.
When I run Swift pair, Marauder displays an exception then reboots.
```
Guru Meditation Error: Core 0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core 0 register dump:
PC : 0x00000000
PS : 0x00060830
[...]
```
I hope it can be worked out :)
Thanks!

wh1ter0z, 2023-11-19:
Hmmm 

I fixed but i get new crash , i defined an array with 3 element 

{"S1","s2","s3"}

Then i put it in packet 

When its on attack mode 

When element being on s3 device will crash ... when it is s1 or s2 is ok without error 

Packet kitchen (apple, android, samsungs.., toy vibrator, windows) spammer is work fine and without crash all of element without crash

coded-with-claws, 2023-11-28:
Hi, in case it can help to analyze, here is the full log of the crash on my device:
```
==================================
> #swiftpair
Starting Swiftpair Spam attack. Stop with stopscan
> Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x801176f1  A1      : 0x3ffe6f60  
A2      : 0x3ffd169c  A3      : 0x00000000  A4      : 0x00000014  A5      : 0x00000000  
A6      : 0x3ffe7020  A7      : 0x80000001  A8      : 0x80081435  A9      : 0x3ffe6f30  
A10     : 0x3ffd169c  A11     : 0x00000000  A12     : 0x00060023  A13     : 0x80000000  
A14     : 0x007bfa08  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400920a0  LEND    : 0x400920b6  LCOUNT  : 0xffffffff  

Backtrace: 0xfffffffd:0x3ffe6f60 0x401176ee:0x3ffe6f80

ELF file SHA256: dbeb52bab7ba2615

Rebooting...
```

wh1ter0z, 2023-11-29:
> Hi, in case it can help to analyze, here is the full log of the crash on my device:
> ```
> ==================================
> Starting Swiftpair Spam attack. Stop with stopscan
>
> Core  0 register dump: [...]

Error from storing string of ble device name in packets 
I fixed it .... 

You have to use ` memcpy ` in getSwiftadvertisementdata func for storing string in packet ....

coded-with-claws, 2023-11-29:
Thanks I'll try that :+1: