justcallmekoko/ESP32Marauder #396: Screen orientation touch inverted on (AP Select,EAPOL/PMKID scan/packet monitor screens)

Wesley5n1p35, 2023-11-18:
**Describe the bug**
All screens in portrait mode work as expected. When the screen flips to Landscape mode 90 Degrees clockwise the touch input acts as though it flipped -90 Degrees Counter-Clockwise.

**To Reproduce**
Steps to reproduce the behavior:
1. Firmware from source (esp32wroom - iLi9341)
2. Touch screen normally in all menus
3. When going to AP Select,EAPOL/PMKID scan/packet monitor screens the touch input is inverted.

**Expected behavior**
The touch input should not be inverted.

**Video Attached**

https://github.com/justcallmekoko/ESP32Marauder/assets/47337218/cb6b173f-2c6d-49e6-bdd4-e618ccc00fca

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.13.4]
 - Hardware version: [ESP32Wroom/iLi9341] 

**Desktop (please complete the following information if applicable):**
 - OS: [WIndows 11/Ardunio 1.8.19]

SSnoody, 2024-08-16:
I have the same issue but with all the marauder in general, i cant use the menu because the touch is inverted

Asmodeux2, 2024-09-17:
Это решается в библиотеке TFT_eSPI в файле User_Settings.h. Есть отдельная строчка с инвестированием True or False