justcallmekoko/ESP32Marauder #402: 'Evil Portal' options inside Marauder apps flipper zero not working

datadotlog, 2023-11-22:
This not go ahead from "Setting HTML"

markjsc, 2024-01-01:
Happy New Year!

I think this one needs a bit more detail. I'll add my observations.

I can't seem to get the HTML file to be properly set when using the Start command.

It works fine with `sethtmlstr` - which is executed from the Flipper app's "Load Evil Portal HTML file" command. But using "sethtml" or "start" (with the "-w" arg) never seems to find the file. This also happens for the ap.config.txt file.

My files are on my SD card at /app_data/marauder/. For example /app_data/marauder/ap.config.txt, or /app_data/marauder/html/Google_Realistic_V2.html.

Any advice on how to get the command to properly recognize the files?

@justcallmekoko

markjsc, 2024-01-01:
> Happy New Year!
> 
> I think this one needs a bit more detail. I'll add my observations.
> 
> From the [ESP32] WiFi Marauder app (v0.6.6), I can't seem to get the HTML file or the ap.config.txt to be properly set when using the Start command. I get the following:
> 
> > Could not find /ap.config.txt. Use stopscan...
> 
> or
> 
> > Could not find Google_Realistic_V2.html. Use stopscan...
> 
> It works fine with `sethtmlstr` - which is executed from the Flipper app's "Load Evil Portal HTML file" command. But using "sethtml" or "start" (with the "-w" arg) never seems to find the file. This also happens for the ap.config.txt file.
> 
> My files are on my SD card at /app_data/marauder/. For example /app_data/marauder/ap.config.txt, or /app_data/marauder/html/Google_Realistic_V2.html.
> 
> Any advice on how to get the command to properly recognize the files?
> 
> I'm using Unleashed Firmware (0.67), Official Wifi Dev Board with Marauader v2.5 firmware. (Please let me know if you need any more details to troubleshoot.)
> 
> @justcallmekoko

Replying to my own post. I was able to get things working as expected using [Xtreme Firmware](https://flipper-xtre.me).

I'm sure the issue still exists under the conditions in my previous post, but I'm glad I was able to work around it using a different firmware.

boobaloop, 2024-01-22:
I was able to get this to work by switching my 64gb kingston microSDXC sd card with a 16gb microSDHC sd card

How big are your SDs? Just switch to a 16gb one

markjsc, 2024-01-22:
> I was able to get this to work by switching my 64gb kingston microSDXC sd card with a 16gb microSDHC sd card
> 
> How big are your SDs? Just switch to a 16gb one

Oooo - that's a good suggestion! I didn't have any micro SD cards around when I ordered the Flipper Zero, so I ordered a multi-pack with 2x64GB. I realize now that it's WAY overkill!

I'll track down a smaller card and see if that fixes it for me.

Thanks for the suggestion!!

boobaloop, 2024-01-22:
I am using a 16gb A-DATA microSDHC UHS-I class 10 and it now shows the AP working perfectly, I was able to test the capturing of credentials and all I did was switch to this card from my 64gb one

markjsc, 2024-01-23:
> I am using a 16gb A-DATA microSDHC UHS-I class 10 and it now shows the AP working perfectly, I was able to test the capturing of credentials and all I did was switch to this card from my 64gb one

Nice! I ordered a 16GB card yesterday (should arrive later this week). I'll post back once I verify the solution.

markjsc, 2024-01-27:
> I am using a 16gb A-DATA microSDHC UHS-I class 10 and it now shows the AP working perfectly, I was able to test the capturing of credentials and all I did was switch to this card from my 64gb one

Following-up. The 16GB card didn't fix the issue with the Marauder app not being able to locate the `ap.config.txt` file.

I can get everything working as expected with the Evil Portal app, however.

Here's what I'm seeing:
- The Evil Portal app uses the `ap.config.txt` file inside the `/apps_data/evil_portal` folder.
- The Evil Portal app looks for HTML files in the  `/apps_data/evil_portal/html` folder.
- The Marauder app does NOT find the `ap.config.txt` file in the following folders:
    - `/`
    - `/apps_data/`
    - `/apps_data/marauder`
- If I run the Evil Portal app once (allowing it to use the `ap.config.txt` from the `/apps_data/evil_portal` folder), then run the Marauder app, the `ap.config.txt` from Evil Portal will be used!

I spent a bit of time trying to debug the Marauder app to determine why the paths were working this way. Sadly, I'm not quite familiar with how things work in C to have made any headway. (However, if you need anything debugged in C# or JavaScript, I'm your guy!)

**So my workaround is pretty simple - use the Evil Portal app for Evil Portal, and use Marauder for everything else.** 

Since I'm using Xtreme firmware, I've customized the menu to include both apps as top-level items:
[image]

trueVinton, 2024-07-11:
Hi guys!

From [the wiki](https://github.com/justcallmekoko/ESP32Marauder/wiki/evilportal), 

> The access point name can be set by the following priority list:
> 1. The first SSID in the list of [SSIDs](https://github.com/justcallmekoko/ESP32Marauder/wiki/ssid)
> 2. The first instance of a "selected" AP in the list of [APs](https://github.com/justcallmekoko/ESP32Marauder/wiki/scanap)
> 3. From /ap.config.txt in the SD card attached to your ESP32

Step 3 is not currently working, but we can still use 1 or 2.
It's easy to do method 2, which creates an evil twin of the existing network within your device range:

1. scan for AP
2. list AP and look for the desired ssid
3. select AP from the list
4. Click "Load Evil Portal HTML file" and select the file. It's using the path `/apps_data/marauder/html`5,  Start evil portal

As a result, you are hosting the evil twin!
POC screenshot:
[image]

closed 2025-09-13 by github-actions[bot]