justcallmekoko/ESP32Marauder #412: SPIFFS Filesystem support instead of SD Card

ssshake, 2023-11-28:
Instead of having to solder an SD card reader to my WIFI Dev board in order to use evil portal, could a change in firmware be made to support a spiffs file system? 

I have done this myself with an ESP8266. You can carve out a portion of the flash memory to be firmware and another portion to be storage using the SPIFFS filesystem. 

Considering it's just trying to host a txt file and an html file an SD card of any size is over kill for this application. 

It would also be much more user friendly because people who do not feel comfortable modifying hardware can still accomplish their software goals.

This will complicate flashing a little bit because in my experience you need to flash the firmware in one memory location followed by flashing the filesystem in another location.

littlefs is another option, there are pros and cons to both depending on how you look at it but I'm not experienced enough to say.

Dowan35, 2023-11-29:
very good idea, still waiting for the internal sd card flipper update tho

IAmOrion, 2023-12-09:
I believe SPIFFS is no longer recommended, in favour of LitlleFS that you mentioned

Ultra980, 2026-02-04:
still relevant