justcallmekoko/ESP32Marauder #416: Is possible working on ESP32 S3?

b4k0s, 2023-12-02:
Is this possible?

mirkokral, 2023-12-02:
When custom compiling, it IS possible. I use my ESP32-S3-DEV-KIT-N8R8 on a perfboard with a display and all works fine.

b4k0s, 2023-12-02:
Ahhh ok ok! i have a LilyGo Esp32-s3 and i wanted to try Marauder on It. You think about compiling for that board? thanks for your answer!

XANKXANK, 2023-12-02:
i have seeed esp32c3 and it works fine with marauder

alisdairg, 2023-12-02:
> ... the Esp32-s3 Just dont Power UP, keeps doing that sound connected/disconnected device on windows.

That sounds more like a cheap USB cable not providing enough power. I've seen that loads of times.

b4k0s, 2023-12-02:
> > ... the Esp32-s3 Just dont Power UP, keeps doing that sound connected/disconnected device on windows.
> 
> That sounds more like a cheap USB cable not providing enough power. I've seen that loads of times. 
> 

Its a original Xiaomi cable. Btw my Esp32-s3 Works Fine running NerdMiner V2 (and i have a battery). So i think its not the cable.[image]

b4k0s, 2023-12-05:
any help?

nysot94, 2023-12-13:
> [image]
> 
> 

How you do that?

b4k0s, 2023-12-14:
> > ![IMG-20231203-WA0022.jpeg](https://private-user-images.githubusercontent.com/152666652/287542030-c4f202d5-cd9f-4e91-ac95-2f8aa092b99c.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1NzMxOTAsIm5iZiI6MTcwMjU3Mjg5MCwicGF0aCI6Ii8xNTI2NjY2NTIvMjg3NTQyMDMwLWM0ZjIwMmQ1LWNkOWYtNGU5MS1hYzk1LTJmOGFhMDkyYjk5Yy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMTRUMTY1NDUwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmE4ODMwODU0ZmZlMWVjMGE4NDk5ZWUwNGUxNGUzY2ZhMDUzN2I1NDU1M2QxOTQ3YmIwMjk3Y2MxYTAzZWZjNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.KywfLTVvLXwR8cITouTxGq5DPTdj9KiCgekcW3g1KjY)
> 
> How you do that?

i used the Dirty Flasher.

nysot94, 2023-12-14:
I got my Lilygo s3 today, I try but don't work

nysot94, 2023-12-14:
There is some YouTube tutorial? I'm noob...

gousdail, 2024-03-14:
have you managed to get marauder to work on your LilyGo Esp32-s3 ?

b4k0s, 2024-03-15:
> have you managed to get marauder to work on your LilyGo Esp32-s3 ?

yes, but its just a standalone... needs to connect via terminal to use commands on it.

avipars, 2024-12-24:
Can you share the bin file?

avipars, 2024-12-24:
> When custom compiling, it IS possible. I use my ESP32-S3-DEV-KIT-N8R8 on a perfboard with a display and all works fine.

@mirkokral

mirkokral, 2025-03-13:
> > When custom compiling, it IS possible. I use my ESP32-S3-DEV-KIT-N8R8 on a perfboard with a display and all works fine.
> 
> [@mirkokral](https://github.com/mirkokral)

 I am sorry, but I no longer have the bin file. It was possibly deleted during my home folder cleanup.

closed 2025-09-13 by github-actions[bot]